miguelbalboa/rfid #213: PCD_Authenticate different keys doesn't work

kajikoffo, 2016-05-25:
Hi,
my problem is that we use Mifare 1K for building access system and to increase security we write some secret into sector 16 of the card (block 60) and change the key A in it's trailer (block 63). Then when reading we read block 60 with this custom key for access. If the card is 'modified', it works OK, but I want to go deeper and report if card has right ID (block 0) but non-modified sector 16, which might be an unsuccessful cloning attempt to report. So I try to auth for sector 16 using our key and in case of failure I retry with factory key, but I never get success for either. I studied MFRC522.h if there's some routine to tell card/reader "sorry, used wrong key, let's retry", but seems there's none. I'm not a great programmer, just self-learning from examples on the web (that debug mess is just temporary). Any help or explanation appreciated. 

I use one function to read block 0 (non-secure) and block 60 (secure).

Code piece:

```
int readBlock(int blockNumber, byte arrayAddress[], bool secure){

...

bool secure_no_success = 0;
  byte status;
  if (DEBUG){
    Serial.print("  --> Reading on reader ");
    Serial.print(reader);
  }
  if (secure == 0)
    status = mfrc522[reader].PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &key, &(mfrc522[reader].uid));
  if (secure == 1){
    if (DEBUG) Serial.println("  --> TRY READ SECURE..  ");
    status = mfrc522[reader].PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &key_sec, &(mfrc522[reader].uid));
  }
  
  if (status != MFRC522::STATUS_OK) {
    if (DEBUG) Serial.println("  --> PCD_Authenticate() failed (read): ");
    if (DEBUG) Serial.print("  --> Status: ");
    if (DEBUG) Serial.println(status);
    secure_no_success = 1;
  }

  if (secure == 1 && secure_no_success == 1){                                                         
    if (DEBUG) Serial.println("  --> ReAuth TRY NOT SECURE.. ");
    status = mfrc522[reader].PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &key, &(mfrc522[reader].uid));
    if (status != MFRC522::STATUS_OK) {                                                             
      if (DEBUG) Serial.println("  --> No success.. ");
      if (DEBUG) Serial.print("  --> Status: ");
      if (DEBUG) Serial.println(status);
      return 3;                                                                                     
    }
    if (status == MFRC522::STATUS_OK) {                                       
      if (DEBUG) Serial.println("  --> Card is a clone!");
    }
  }

   byte buffersize = 18;
  status = mfrc522[reader].MIFARE_Read(blockNumber, arrayAddress, &buffersize);
  if (status != MFRC522::STATUS_OK) {                                                                 
    if (DEBUG) Serial.println("  --> MIFARE_read() failed: ");
    return 4;                                                                                         
  }
````

all I always get is:

```
  --> Reading on reader '0'!
  --> TRY READ SECURE..  
  --> PCD_Authenticate() failed (read): 
  --> Status: 3
  --> ReAuth TRY NOT SECURE.. 
  --> No success :/.. 
  --> Status: 3
```

I tried to add commands mfrc522[reader].PICC_HaltA() and mfrc522[reader].PCD_StopCrypto1() before retry, I tried 5V/3.3V power for reader.. Seems like once I try our modified key when the card has the factory one set I can't retry until I get the card from range of reader and approach it again.

simbadMarino, 2016-06-21:
Try initializing the driver again with: 
 _mfrc522.PCD_Init();        // Init MFRC522 card_
before reading the card,it worked for me to avoid re-inserting the card again. Please let me know if you need an example

Rotzbua, 2016-08-25:
@kajikoffo  Hi. Did you found a solution? Will you share your solution? Or can I close this issue?

kajikoffo, 2016-08-26:
Hi, sorry for not sharing, I got swallowed by work. Approach of @simbadMarino simbadMarino works, I tried it in a 'for' cycle using different key every time and if you reinit the reader it instantly looks for a card. Here's a piece of code used (bt_f is 6th byte of Key-A) as it worked for me:
```
...
for (byte bt_f=0; bt_f<255; bt_f++){
  key.keyByte[5] = bt_f;
  mfrc522.PICC_HaltA();
  mfrc522.PCD_StopCrypto1();
  mfrc522.PCD_Init();
    if (mfrc522.PICC_IsNewCardPresent()){
    if (mfrc522.PICC_ReadCardSerial()){
    byte status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, 3, &key, &(mfrc522.uid));
    if (status != MFRC522::STATUS_OK) {
    }
    if (status == MFRC522::STATUS_OK) {
...
```
It runs the cycle, trying all the 255 possibilities takes under one minute, so it's good if you want to try several keys, but it would be terribly slow if you'd like to brute-force try to guess it (which is given by RC522 chip I guess).
Thanks guys for your help.

K.

Rotzbua, 2016-08-26:
thanks for sharing

closed 2016-08-26 by Rotzbua