miguelbalboa/rfid #214: MFRC522.h problem

BSankey, 2016-05-27:
My Arduino will not compile the MFRC522.h file.
Could you please help?
The errors are as follows
Arduino: 1.0.6 (Windows NT (unknown)), Board: "Arduino Uno"
C:\Program Files\Arduino\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=106 -IC:\Program Files\Arduino\hardware\arduino\cores\arduino -IC:\Program Files\Arduino\hardware\arduino\variants\standard -IC:\Program Files\Arduino\libraries\SPI -IC:\Program Files\Arduino\libraries\MFRC522 C:\Users\Brian\AppData\Local\Temp\build3464010330650456565.tmp\DumpInfo.cpp -o C:\Users\Brian\AppData\Local\Temp\build3464010330650456565.tmp\DumpInfo.cpp.o 

In file included from DumpInfo.ino:37:
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:87: warning: only initialized variables can be placed into program memory area
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:99: warning: only initialized variables can be placed into program memory area
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:111: warning: only initialized variables can be placed into program memory area
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:123: warning: only initialized variables can be placed into program memory area
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:276: error: use of enum 'PICC_Type' without previous declaration
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:276: error: expected primary-expression before '{' token
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:276: error: expected ';' before '{' token
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:291: error: use of enum 'StatusCode' without previous declaration
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:291: error: expected primary-expression before '{' token
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:291: error: expected ';' before '{' token
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:337: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:354: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:355: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:356: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:357: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:358: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:359: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:360: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:365: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:367: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:368: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:369: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:370: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:371: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:372: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:373: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:374: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:375: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:376: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:381: error: 'StatusCode' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:384: error: expected ';' before '(' token
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:385: error: 'PICC_Type' does not name a type
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:388: error: expected ';' before '(' token
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:394: error: 'PICC_Type' has not been declared
C:\Program Files\Arduino\libraries\MFRC522/MFRC522.h:413: error: 'StatusCode' does not name a type
DumpInfo.ino: In function 'void setup()':
DumpInfo.ino:50: warning: only initialized variables can be placed into program memory area

Rotzbua, 2016-05-28:
dublicate  #204

Rotzbua, 2016-05-28:
duplicate to #204

closed 2016-06-17 by miguelbalboa