miguelbalboa/rfid #218: RFID dumpinfo export to excel

suisho11, 2016-06-18:
Hi, 

I am new to software and RFID. Could you please help, how can i export the dumpinfo into excel at the same time convert the hex to String?

Thank you

simbadMarino, 2016-06-21:
This library should only be taken as a driver, you have to create a Java, C#,Android or some other language app to read the serial port inputs and create the file for you. I think to do it using Arduino is not impossible but for sure more complicated than a PC/Smartphone approach. 
Is your application going to run standalone? If so then probably yo want to create a .csv file in a SD card to be processed later in a PC

Rotzbua, 2016-08-25:
@suisho11 Hi. Did you found a solution? Maybe share your knowledge.
Otherwise I will close this issue.

Rotzbua, 2016-09-24:
no response -> close

closed 2016-09-24 by Rotzbua