miguelbalboa/rfid #220: NFC - Help to exchange data between RC522 & Android hce

romainhry, 2016-06-21:
I will explain my project :

I have my RC522 and a door connected on my Arduino UNO.

I can currently open the door with a MIFARE classic.

But now I want to open it with my Android smartphone, this is why I develop a HCE applet to accept the good APDU with the selected AID, then my phone will transfer the data in order to open the door.

But the problem is :

I don't know how to send an APDU command with my Arduino using the RC522.

Currently, for my MIFARE Cards, I use the https://github.com/miguelbalboa/rfid library.

My test code :

```
byte selectApdu[] = { 
  0x00, /* CLA */
  0xA4, /* INS */
  0x04, /* P1  */
  0x00, /* P2  */
  0x05, /* Length of AID  */
  0xF2, 0x22, 0x22, 0x22, 0x22,
};
byte * backData = (byte *)malloc(16*sizeof(byte));
byte * dataLen = (byte *)16;

status = mfrc522.PCD_TransceiveData(selectApdu,10,backData,dataLen,NULL,0,false);
if ( status != MFRC522::STATUS_OK) {
    Serial.print(F("PCD_TransceiveData() failed: "));
    Serial.println(mfrc522.GetStatusCodeName(status));
    return;
}
else
{
  Serial.println(F("PICC_TransceiveData() success "));
}
```
Neverless, it doesn't work ( "Timeout in communication"), and I slowly need to think that the RC522 is not compatible... Confirm me this plz, and tell me why

Rotzbua, 2016-06-22:
Modern NFC has nothing to do with Mifare classic.
Old smartphones have compatibility mode for MFclassic but modern smartphones dropped it because MFclassic is too old.

romainhry, 2016-06-22:
This is supposed to confirm that RC522 is not compatible with HCE ?

Rotzbua, 2016-06-22:
Yes, I wrote this already in the readme: (I found years ago a project with mention this, sorry cant remember where)
```
My mobile phone doesn't recognize the MFRC522 or my MFRC522 can't read data from other MFRC522

    Card simmulation is not supported.
    Communication with mobile phones is not supported.
    Peer to peer communication is not supported.
```
Use a newer chip with more features, readme:
```
If hardware: buy a more expensive like PN532 (supports NFC and many more, but costs about $15)

romainhry, 2016-06-22:
Yes I saw this features. Nevertheless I didn't understand why it doesn't work if my phone is ISO 14443-4 type A compliance.
I bought (SeedStudio NFC Grove - pn532) this morning thx for the answer.

Rotzbua, 2016-06-22:
Ok. Can you read mf classic token?
Maybe test increasing the timeout limit, it is somewhere hardcoded in the lib.
I tested it with my smart phone and I can not read mf classic tokens. (tested several apps)

romainhry, 2016-06-22:
Yes I can read mf classic token, nevertheless the card has not the same reactions even if the data are the same (even key A and key B and access bytes). Only block 0 is different.

Rotzbua, 2016-06-22:
Got it:
For communication it seems that feature `Peer-to-peer (ISO/IEC 18092)` must be supported. Maybe with `Card emulation` it is possible, but MFRC522 has nothing of that.

Source:
http://www.nxp.com/documents/leaflet/939775017564.pdf
https://web.archive.org/web/20151210045625/http://www.nxp.com/documents/leaflet/939775017564.pdf

commit e9003130 (2016-08-25): upd readme

add note for compatible ide
closes #200
closes #204
closes #205
closes #212
upd readme
closes #220
closes #36
extend gitignore with ide clion data
closed 2016-08-25 by Rotzbua