miguelbalboa/rfid #221: using UART peripheral instead of SPI for comminucating between MCU & PCD

Dizgah, 2016-07-03:
Hi,
I'm trying to change MCU <-> PCD connection to SPI.
based on MFRC522's datasheet ,every thing is setted & well run(for writing to PCD's register when I write address and then data after it,PCD returning me register address) .
But problem is when i want to sense presence of PICC , in PCD_CommunicateWithPICC function line425 loop runned 2000 time which i think mean PCD does not return appropriate value,
has any body experience to using this chip with UART periph or any suggestion ?
(every thing is ok with SPI)
WBR.

Dizgah, 2016-07-04:
OK,Address byte framing in UART mode is different with SPI,although I modify that but still does not work(PCD chip version works corectly)

Rotzbua, 2016-08-25:
@Dizgah Hi. Did you found a solution? Or a more detailed bug description?
Otherwise I will close this issue.

Dizgah, 2016-09-05:
Hi ,
Problem was related with address framing differences in UART & SPI mode.By changing body of address byte to the UART mode it will be works fine,
WBR.

closed 2016-09-05 by Rotzbua

Rotzbua, 2017-01-29:
@Dizgah Did you implemented UART? Maybe you can give @cyberjuputer some hints at #281 . thanks.