miguelbalboa/rfid #223: trouble writing in the upper blocks to 4 in the 1K Mifare card

sistemasneytorres, 2016-07-07:
Hello everyone,

I am writing to ask for your help, because I'm writing a code to read and write Mifare 1K cards, using a mega arduino and rfid-rc522 module. The problem that I present is that only managed to read and write in blocks 1 and 2 of the card (Autenticándome with the key A), when I try to write in the following blocks (from block 4) I get an error message NAK. And I tried to authenticate with the keys A and B but I get the same error. I used the example code available https://github.com/miguelbalboa/rfid/blob/master/examples/rfid_write_personal_data/rfid_write_personal_data.ino.

I honestly do not know if the card model only allows write only in blocks 1 and 2, or there really is a problem with the code that does not allow write in the following blocks.

Any help or information that allows me to understand what is happening will be enormously grateful.

Happy day to everyone,

simbadMarino, 2016-07-10:
Hola Torres,
Check my repository here:
https://github.com/simbadMarino/CardIDFlashTool
I've created a block by block reader and writer. I0ve only tested it with 4K tags, but maybe it works also for 1K.

Let me know if it works! See ya!

Rotzbua, 2016-08-25:
@sistemasneytorres Did you solved your issues? Will you share your solution? Can I close it?

sistemasneytorres, 2016-08-26:
Hi,

thanks for your help, honestly I did not continue with that project so I
could not test your solution.

You can close the case, and whe I resume the project again, I will let you
know if I having the same trouble.

Regards!!!!

closed 2016-08-27 by Rotzbua