miguelbalboa/rfid #224: Porting this library for LUA and esp8266 ?

Netoperz, 2016-07-09:
Hello ones again,
I'm working on my private project to get UID of tag using rc522 and ESP8266 but... with lua firmware onboard, just like nodeMCU boards and so...

This is not an issue thow, but just a question does anyone already worked out the solution ?
As LUA has got much more to offer than standard Arduino IDE in terms of processing power and timers it would be superb to have access to NFC when using LUA, it is hard to find a solution online, and many i could help in developing that kind of library port as I am working on the project for my self ?

Just send me a message or post here if anyone has got any code that already works with LUA language.

Cheers.

M.

Rotzbua, 2016-07-11:
First: NFC is not supported.

Second: You did not understand correctly what the Arduino ide is. It is simply a c/c++ compiler with upload. LUA and C++ are totally different programming languages.
This library is basically a driver. Programming a driver in lua is in my opinion not a good idea, because it do not have the same performance and timing as c++.

Third: I think you can import and use c++ libraries as external source with lua.. ask a big search engine ;)

Best

Rotzbua, 2016-08-25:
@Netoperz Hi. Did you found a solution? Maybe share your knowledge.
Otherwise I will close this issue.

piebat, 2016-09-05:
Hi. I'm working to try use the NFRC522 with LUA and a WROOM-02 module. Using the spi module of NODEMCU I try at least read the ID of card. Unfortunately at moment my code can perfectly read all the registers of PCD, but seems that it cannot write on them. So doesn't work. I'm studying signal of spi and timing, they seems compatibles with data-sheet, but any change I try to make to any register, seems don't apply and I read always the default value. I'll keep you update.

Netoperz, 2016-09-05:
@piebat 
I'm sorry I was offline due to the holidays and a lot of work with my projects.
It's good that someone else is looking/working on solution too.
I will share some info also as fast as i will test the code.

@Rotzbua 

Thx for reply, I Know that LUA and Arduino IDE are different :) but as many may notice people using Ar IDE also use LUA or want to, because it has more flexibility in a way is baby "better" for production... depending on needs...
So asking "does anybody know something about...?" in Arduino community is a good step to find someone that struggles the same fears. 👍 

Right now we are working on other hardware support,(one device have to have Bluetooth, WI-FI , OLED, RGB LED, Joystick , accelerometer, speaker, and vibration unit. and .... NFC

Right now we have got support on almost everything on the list, with small glitches to solve :) last is NFC, we will start work next month I think. If I will find some solution than  I will share my knowledge as always, It is very very nice to see that someone else can build things using the knowledge that some other has created.

Let's let this tread open for a while.

If there is someone working on using RC522 on LUA to get at least UUID , than I'm open for suggestions 

Cheers

Rotzbua, 2016-09-06:
@Netoperz Please note that the mfrc522 do **not support nfc**, nfc is a different newer iso standard. They are not compatible.

Rotzbua, 2017-01-26:
Can be reopened if somebody wants to work on it.

closed 2017-01-26 by Rotzbua

goodshileidev, 2018-03-05:
@Netoperz

Have you resolved your problem ? I have a same need with you to use RC522 with lua, and I found the solution below, but it doesn't work. Can you have something on this topic?

https://github.com/capella-ben/LUA_RC522