miguelbalboa/rfid #226: Working with Arduino 101

prazek, 2016-07-19:
Does anyone know if this is compatible with Arduino 101? Sketch was written sucessfully, but it doesn't seem to work.

Rotzbua, 2016-07-20:
Read old issues #225

prazek, 2016-07-20:
Thanks for the comment that doesn't add anything new to the disscussion. I have read the documentation and I know what boards are officially supported. If Arduino 101 is not one of them then I would like to know why and what should I fix to make it work so everyone could use it.

Rotzbua, 2016-07-20:
Sorry, I read too fast. Arduino 101 != Arduino IDE 1.01, sorry for my mistake.
I think it should work. Maybe replace unfixed integers with fixed integers.

prazek, 2016-07-20:
I will firstly check if I solted RC522 properly using older version of arduino. If I will make it work I will send patch adding support for 101 or fix the issue. Thanks!

Rotzbua, 2016-08-25:
@prazek any progress? does it work with board `Arduino 101`?

prazek, 2016-08-25:
I managed to make it work on older arduino by replacing the RFC522. Haven't tried with 101, but I will check if it works. If so I will update documentation.

prazek, 2016-09-11:
Ok, I finally got tome ti test it.
https://github.com/miguelbalboa/rfid/pull/246
everything works.

closed 2016-09-11 by prazek