miguelbalboa/rfid #227: Multiple readers more than 9

cybersersupremo, 2016-08-01:
Hi, im having a lot of problems with connectin 9 readers.
For the moment i have working 9 but sometimes 1 dont response, not always the same , but only if i connect to the computer.
But if i connect to the power supply,only works 4 or 5 ,but no more.

i tried a lot of things, i want to try tomorrow to slow down the SPI_CLOCK, where i can t o put the CLOCK DIVIDER, i can use in  my sketch after the init of the SPI. or  do i have to change in the library?

Any more ideas?
Regards

Rotzbua, 2016-08-03:
9 devices are a lot. I think you operate on the physical maximum of supply and signal.

cybersersupremo, 2016-08-03:
And what can i do? can i use an external power supply or something?

Now i have working perfectly with usb, but if i use a power supply does not work, possible noise?
any idea for solve it? 
thanks

Date: Wed, 3 Aug 2016 15:16:08 -0700

Rotzbua, 2016-08-04:
No, idea. It is not a software problem, so please close this issue.
Ask https://forum.arduino.cc/ ..

closed 2016-08-25 by Rotzbua