miguelbalboa/rfid #228: How to change key of MIFARE MF1ICS50 NFC tag ?

AyushMarsian, 2016-08-04:
Hello,

Our company is going to **use NFC MIFARE MF1ICS50 with RC-522 Reader/writer module with ARDUINO**.
I can access the card but i don't get how to modify it's parameter,like **how to change Key A and Key B of each sector, how to use and change access bits** ?

I was use this function "**mfrc522.MIFARE_SetAccessBits**" but i can't.

can you tell me how to use this function for change KeyA and KeyB,and tutorial for learn this card and module.

Please help me as soon as possible

Thank you

LoyalNine1487, 2016-08-10:
Don't use the key option, Its been broken for years and so insecure. If your doing it for work you should use a newer card type and library.

TomHyhlik, 2016-10-10:
have the same problem, please help

Lube, 2016-10-27:
@LoyalNine1487 Which Key option is deprecated? Which other mifare library do you recommed?

Rotzbua, 2017-04-07:
LoyalNine1487 means that you should not use the key option for important data, because the crypto is broken. This means do *not* use it for commercial stuff!!

The function `mfrc522.MIFARE_SetAccessBits` do not modify anything on the hardware. It is only a helper function to calculate the bit pattern which have to be set.

> Calculates the bit pattern needed for the specified access bits. In the [C1 C2 C3] tuples C1 is MSB (=4) and C3 is LSB (=1).

@AyushMarsian Can this issue closed?

closed 2017-04-21 by Rotzbua

AyushMarsian, 2017-06-10:
Hello Friends 

Thanks **Rotzbua** for this library, thanks friends for your replay.

I changed encryption key of my MIFARE MF1ICS50 card by modifying example **"MifareClassicValueBlock"**.

But now I'm facing another issue, Normally my system is works well but if some time we remove card from RC522 while it's reading datablock then it's gets hanged and then it will not recover back.

I have tried soft reset by library but it's not work.

what i have to do for this ?
Currently I'm continue with switch off and on power but if you have solution then please help me.

AlbertMN, 2018-01-10:
Still a valid question ^

AyushMarsian, 2018-01-10:
@AlbertMN  Sorry but i don't understand what you are asking for ?

AlbertMN, 2018-01-10:
@AyushMarsian Wrong forum, sorry :)