miguelbalboa/rfid #233: Philips Mifare 1 S50 tag compatible?

stevekramer99, 2016-08-22:
Hello I got this reader kit on Ebay:

http://www.ebay.com/itm/371104303227

The reader has no response to the keychain tag but responds to the card.  The description says the tag has the following IC:  "Philips Mifare 1 S50"  Should the library work with this tag?

Thanks,
Stephen

simbadMarino, 2016-08-22:
It should work, recheck wiring and try not to use a protoboard,  always use a direct wire soldering. 
Good.luck!

stevekramer99, 2016-08-22:
The reader works fine with the card, but does not respond at all to the
keychain tag.  I don't touch the wiring between reading them just scan each
one sequentially.  There is no response to the keychain fob.  It must be
faulty, please send a replacement correctly working tag.

stevekramer99, 2016-08-22:
I AM SORRY, I SENT THIS TO THE WRONG PERSON!  I was supposed to send this
to the ebay seller!

Thanks for your help.

Rotzbua, 2017-01-15:
@stevekramer99 Did it work with the Philips Mifare 1 S50 ?

closed 2017-01-30 by Rotzbua