miguelbalboa/rfid #234: Overwrite Trailer Block 7 ?

postolache, 2016-08-23:
By mistake I've wrote 0x00 on all sector 1, including Trailer Block 7, and now I get 
`   1      7  PCD_Authenticate() failed: Timeout in communication.
`

Is any way to reset card to factory values or to gain back access to that sector ?

TomHyhlik, 2016-10-10:
Yeah I rewrited Trailer Block 7 too, but but with AAAAAAAAAA on the end of the block where the key is located and I didn't find out how to rewrite it back to FFFFFFFFFFFFFF. Authenticaion failed.

dangear, 2017-02-16:
Same problem and same blocks (when I try to clear this sectors) . But can read UID and write to 8-63 blocks.

Rotzbua, 2017-02-16:
Guys/Girls this issue is still useless if there is no information about reader/tag type/firmware/revision.

dangear, 2017-02-16:
Sorry!
RFID-RC522
Firmware Version: 0x92 = v2.0
Card SAK: 08
PICC type: MIFARE 1KB
it worked before I wanted to erase blocks. And I think I broke some data in block#7

Rotzbua, 2017-02-16:
As always, the data sheet has an answer:

> Remark: 
> With each memory access the internal logic verifies the format of the access conditions. If it detects a format violation the whole sector is irreversibly blocked

Conclusion: every 3rd block (3,7,11,15,..) is a tailor block which manage the access, if you set it wrong you will brick your PICC. That's the meaning of a access control system.

The data sheet http://www.mouser.com/ds/2/302/MF1S503x-89574.pdf gives an overview where data can be stored and where the key/access control bytes are. See capture: 8.6   Memory organization.

In my opinion it is not a library bug and I think I can close this issue. Any other opinions? @postolache 
@TomHyhlik @dangear

dangear, 2017-02-17:
No more question!)
But maybe in this library you can do (by default) rule: MIFARE_Write() function can write data if number of block not 3,7,11 (and other). And if user want to change access key for section he need to use function like a MIFARE_ChangeSectionKey(section,key)

Rotzbua, 2017-03-01:
@dangear That would be a lot of work to adapt it for all available tags (1k, 4k, mini, clones, ..). A new high-level class with e.g. `writeRaw`, `writeData` and `writeAccess` would be nice. If somebody implements, it can be merged. I have not time for this.

closed 2017-03-01 by Rotzbua