miguelbalboa/rfid #236: Arduino Pro-Mini has loss of power when I connect the ground pin to RC522

dsierra73, 2016-08-30:
Can anyone help? I have soldered and checked every pin. I do get the arduino to work the blinking samples. the moment I connect RC522 it all seems to work and when connected to sketch it appears connected with all the pins... the moment I plug ground from rc522 to arduino ground, it stops working and is not longer found by the computer. there's a message on Sketch that states "Board at COM10 is not available"  Unplug ground and arduino comes alive again.

tuyethoa08041997, 2016-09-05:
I'm using Pro Mini to communicate with RC522, it should work fine.
If you connect everything right and it still not work, maybe your RC522 is dead...
(RC522 use 3.3v as VCC, I burn my RC522 once because of wrong VCC)
(Sorry for my bad English :| )

Rotzbua, 2017-04-20:
@dsierra73 Did you solved your problem? Can you share your solution?

dsierra73, 2017-04-20:
I had to buy a new rc522. Once I got the new one it worked fine. I would say that you should buy it already soldered on. The first one I bought I had to do myself and maybe I burned a circuit or something. The second I bought had all parts already attached and it worked fine.

Rotzbua, 2017-04-21:
Thanks for your feedback :) , hardware of the mfrc522 seems to be always difficult..

closed 2017-04-21 by Rotzbua