miguelbalboa/rfid #240: Connecting RFID via I2C

lifebearups, 2016-09-06:
Hi, I was browsing the MFRC522 datasheet (http://www.nxp.com/documents/data_sheet/MFRC522.pdf) and I saw that it has I2C support. I couldn't find any implementation of it online, or anyone that has performed that. 

Is it possible to communicate with this component via I2c?

tuyethoa08041997, 2016-09-07:
Never see  any implementation before, maybe because I2C's speed is slow...

piebat, 2016-09-07:
the most popular board on internet (and the cheapest) is set for SPI only. I found another with i2c (PCB board was red color), but it was more expansive. In this moment I don't have a link, but you can search on internet. The chip can use UART, SPI or i2c, but the board that I found are not configurable, so generally them are made for one only of the protocol. So I think that if you don't find is because the board are hard to find.

closed 2017-01-16 by Rotzbua

girlpunk, 2022-02-05:
Not sure if you're still looking for a cheaper device to use for adding I2C support, but there are now ones available for around $5

https://shop.m5stack.com/products/rfid-unit-2-ws1850s

commit 0b91afb4 (2017-01-16): Update README.rst

close #111
close #240