miguelbalboa/rfid #243: 2 mfrc522 Readers for different gates

Amorasamy, 2016-09-09:
Can i differentiate between 2 readers as each one only accept special UIDs ? i mean each one works individually as a special gate ?? 
because in "ReadUidMultiReader" example i see that they are like a one Reader with 2 ends (nodes)
Thanks

tuyethoa08041997, 2016-09-09:
Yes you can, but it needs 2 SPI select line and, i don't know, maybe a little change in the library?

Amorasamy, 2016-09-09:
yes 2 SS lines on the same board , but the question is about the library , do it support that or no ? :)

tuyethoa08041997, 2016-09-09:
AFAIK, it is not directly support. Try to do something like this:
```
MFRC522 reader1;
MFRC522 reader2;
```
...

Amorasamy, 2016-09-09:
i'll try that , i hope it work as i like.
 Thanks

DrLou, 2016-09-09:
not yet expert with this stuff... Should we be able to send a parameter to the loaded Arduino code - over either SPI or I2C? E.G. can we send an 'accept this key' parameter to the device?

Amorasamy, 2016-09-09:
You will use SPI protocol with mutual pins between two RFID readers.
what you mean by device ??

lmmeng, 2016-09-11:
@tuyethoa08041997 : Which is the difference between using two readers like
```
MFRC522 reader1;
MFRC522 reader2;
```
and using an array of readers?
```
MFRC522 myReaders[2];
```
In first case you use the names, in the second use the array's index to select the right reader, but they are two different readers.

@Amorasamy: of course you can use the readers independently. The MultiReader  example is just an example to show (how) it works, no need to do different actions for each reader. 
But you can use the reader's index to differentiate the readers (the reader = 0; reader < NR_OF_READERS; reader++  in the example) :
```
  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    // Look for new cards

    if (mfrc522[reader].PICC_IsNewCardPresent() && mfrc522[reader].PICC_ReadCardSerial()) {
```
You should, of course, interpret the read data accordingly with your needs, but it should work.

Amorasamy, 2016-09-11:
Yea @lmmeng you are right ! .if i used reader[1] and reader[2] in any condition it will do what i want , i didn't notice that at first but a friend of mine told me about it.
Thank you for your replying :+1:

lmmeng, 2016-09-11:
You mean reader[0] and reader[1], don't you?

Amorasamy, 2016-09-11:
yup

Rotzbua, 2017-01-19:
@Amorasamy Did you solved your problem? Can you share your experience with us? Otherwise I will close this issue within 14 days. Thanks.

closed 2017-02-17 by Rotzbua