miguelbalboa/rfid #247: Communications problem: Firmware Version: 0x0 = (unknown)

nimgog, 2016-09-13:
Hi!

It seems I have a problem with my MFRC522. 

Problem message prompted when opening serial monitor: 
"Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
Scan PICC to see UID, SAK, type, and data blocks..."

I have found similar threads across the internet, yet none with a proper solution.

I am using an Arduino pro mini (chinese copy: https://goo.gl/ZpXXmv ) atmega 328, 3,3V with a FTDI for programming. Everything is soldered together using dupont lines. 

Troubleshooting done so far:

1. check and recheck all soldering, as well as rechecking the pin layout for not having wired it wrong.
2. Trying RST pin (3) instead of pin 10 for reset. 
3. See if there might be some problem with drivers installed for the Arduino or FTDI. 
4.  Trying uploading different scetches to see if I can get some other reaction.
- which all lead back to not being able to identify the version of the MFRC522. (Note: I have also had 0xFF prompted as unknown version for the same card). 

All of which leads to nothing.

I'd be utmost grateful for some guidance in this.

Best regards,
Nimgog.

Rotzbua, 2017-01-26:
@nimgog Did you solved your problem?

nimgog, 2017-01-26:
It seems as if there was a problem with either the processor or the mfrc522. I haven't ruled either of them out yet as I bought an arduino uno instead of the pro mini and I also went ahead and bought a new mfrc522. This time both from legit retailers. My problem was resolved after that.

Rotzbua, 2017-01-26:
Thanks for your quick answer. I worked with them too. They define the default value of `SS` different than other arduino boards, did you mention this?

nimgog, 2017-01-26:
You mean the pro mini defines the values of ss different? No I didn't mention that. Different how exactly?

Rotzbua, 2017-03-26:
Hi @nimgog , sorry for the delayed answer. It was the `pro micro` where I had some trouble:
SS pin was 17 and not 14 which I has expected.
```c++
static const uint8_t SS = 17;
```
https://github.com/sparkfun/Arduino_Boards/blob/9a2fcd2e8d84046745af742bc12c5169335b9d86/sparkfun/avr/variants/promicro/pins_arduino.h#L109

closed 2017-06-04 by Rotzbua