miguelbalboa/rfid #249: PCD_Authenticate() failed: Timeout in communication.

lal12, 2016-09-20:
Hello,

I am using the lib on the Arduino Leonardo, I now encountered the following problem, when I write something on the rfid chip with the android app NFCTools, I from this point can't dump the chip anymore instead I get the following error:
PCD_Authenticate() failed: Timeout in communication

I could reproduce this with several chips (NXP Mifare Classic 1K)

jackblk, 2016-10-19:
I just started today. I figured out if you use NFCTools, it will format your card into NDEF. This RFID code doesn't work on NDEF format, it only works on blank card.
Read more about these 2 formats here: 
https://learn.adafruit.com/adafruit-pn532-rfid-nfc/ndef
https://learn.adafruit.com/adafruit-pn532-rfid-nfc/mifare

The thing is, the authentication is different between these two. Key A and B from blank card is 0XFF 0XFF 0XFF 0XFF 0XFF 0XFF, while NDEF has different keys.

You can reuse this code: https://github.com/BoolLi/RC522_RFID_NFC_Utilities
It has read/write/format solution for NDEF cards, but you gotta edit it someway. Hope this helps, i'm trying to edit it to make it work for both blank and NDEF cards. Cheers!

Rotzbua, 2016-12-23:
@lal12 Did you solved your problem? Can you share your experience with us? Otherwise I will close this issue within 14 days. Thanks.

closed 2017-01-15 by Rotzbua

cologne86, 2017-01-22:
i ran into the same problems: [https://github.com/BoolLi/RC522_RFID_NFC_Utilities] solved the problem for me(thx to jackblk), i had to modify the cpp to get it compiled. In line 95, 130, 267 and 276 you have to change **byte** status/readStatus to **MFRC522::StatusCode** status/readStatus, because the rfid library has changed the datatype.

Radulepy, 2017-07-13:
@cologne86
I can't find anything in examples, can somebody help me ?
I changed byte -> MFRC522... like you said and can't find it.
Thank you

SolidRacoon, 2022-04-15:
I found a solution for me, idk if it work for you but you can try. I know it's a strange solution, but it works. I download the app mifare classic tool on smartohone. Choose "write tag" -> "factory format" -> "select all" -> "start mapping and format tag". active your nfc option on the phone. Put the card on the back and start mapping. it's reset the data and after u can write data ^^ hope that help <3 (and sorry for my poor english)

KROKODYL4, 2022-06-20:
> I found a solution for me, idk if it work for you but you can try. I know it's a strange solution, but it works. I download the app mifare classic tool on smartohone. Choose "write tag" -> "factory [...]

thank you, it worked!