miguelbalboa/rfid #251: Clone reader error

tib1989, 2016-09-28:
Hi, 

My goal is to copy my first card to my second card wich is brand new. I get some errors:

When I read my first card i get that:

[image]

on my second card i get that 

[image]

but when i try to copy the data from my first card i get that

[image]

Anyone got an idea ?

Thank you

thomaslc66, 2016-10-16:
Hi, same problem here.

For the first timeout in communication you need to keep the RFID card to the reader without moving. 

But for the copy of the card, i have the same problem. I've made some research and they told it can be a problem with the used key ( A or B), but the library is testing both, so I don't know and i'm also searching for a solution

simbadMarino, 2016-10-16:
Hello folks, I had a similar issue with another program, try using authentication only with one key A or B not both. That resolved my problems, I didn't analyzed deep enough to give u a reason why this is happening but it helped me. Good luck! 

⁣

​

Rotzbua, 2016-12-23:
@tib1989 Did you solved your problem? Can you share your experience with us? Otherwise I will close this issue within 14 days. Thanks.

closed 2017-01-15 by Rotzbua