miguelbalboa/rfid #261: How to call reset funtion

BobMcNair, 2016-11-22:
Hi,
I am writing a program to detect when certain cards are present, and also when they are not.
The RC522 reader seems to remember the last card it read. and this doesent clear till a new one is read.
I need to re-set this. I cant get the reset command to work, what is the correct syntax to call 
this??

omersiar, 2017-02-02:
You may want to clear some cache. I do not think there is a reset command to clear buffers or something, I cannot guarantee but UID buffer can be purged like this.

Why do you need it for? Is there any security concerns?

```
for (uint8_t i = 0; i < mfrc522.uid.size; i++) {
    mfrc522.uid.uidByte[i] = 0;
    //Serial.print(mfrc522.uid.uidByte, HEX);
  }
```

Rotzbua, 2017-02-17:
@BobMcNair Did you solved your issue?

closed 2017-03-01 by Rotzbua

illosan, 2021-05-20:
> 
> 
> @BobMcNair Did you solved your issue?

It worked for me. I had a false reading when the card was too far away or passed too quickly. By resetting the value of the last card read, I was able to check the exact reading. Thank you