miguelbalboa/rfid #263: multiple RFID 552 scanners problem

propiro, 2016-11-27:
Hello,
While using this library, im getting really weird results. I must connect 5 scanners to one arduino. My scanners share MISO SCK MOSI and power lines, and when i connect 1 or 2, they work as expected, but when i connect 3,4,5, all of them stop working - only if i disconnect MISO lines of any 3 of them, it'll make other two work.
At this point im clueless, i've tried using voltage of 3.3, 5v, using 5 to 3.3v stabilizers, basically anything i can thing of. I've tried two MEGA arduinos, one leonardo and one UNO, on all of them i have similiar problem.

http://nopaste.linux-dev.org/?1120284
here is code, from multireader example, with added 5 scanners.

simbadMarino, 2016-11-27:
Check voltage drop⁣ in your 3.3v or 5.0v, looks like you are exceeding the maximum current and this is causing malfunction in your setup. 
Good luck!

Rotzbua, 2016-11-27:
maybe duplicate #227

Rotzbua, 2016-11-27:
In general my experience with spi:
- breadboard + long connector cables -> issues
- breadboard + >=2 devices -> issues

Try:
- shorten connection between Arduino and devices
- use a soldered prototype instead of a breadboard

or

- Arduino mega has more than one hardware spi -> use them -> modify library (I have no clue how)

Rotzbua, 2016-12-23:
@propiro Did you solved your problem? Can you share your experience with us? Otherwise I will close this issue within 14 days. Thanks.

propiro, 2016-12-24:
Hello,
Sadly, it was caused by noise on MISO lines, and i was unable to solve it,
using multiplexing or transistor gates. Client i was makign this project
for agreed just to but 5 cheap arduinos, so i hook them together to one
arduino mega that controlled locks, and everything worked.

Best Regards,
A.

Rotzbua, 2016-12-24:
@propiro Cool! Thanks for sharing your knowledge with us. Merry Christmas.

closed 2016-12-24 by Rotzbua