miguelbalboa/rfid #264: PCD_reset goes to infinite loop

zespar, 2016-12-04:
Hi,
 I am unable to initialize mfrc522 when i connect mega to laptop and try to see some logs on console. On further investigation i figured out that **PCD_Reset** goes to infinite loop which calls the function below and the value returned by PCD_ReadRegister is always  **0xFF**. It only happens when serial is connected. 

> byte MFRC522::PCD_ReadRegister (byte reg ) {
	byte value;
	digitalWrite(_chipSelectPin, LOW);			// Select slave
	SPI.transfer(0x80 | (reg & 0x7E));		
	value = SPI.transfer(0);			
	digitalWrite(_chipSelectPin, HIGH);	
	Serial.println(value, HEX);
	return value;
}
 
Thanks

pedroluis5, 2017-03-15:
hello

there is a problem on PCD_init.

change the line 
`pinMode(_resetPowerDownPin, OUTPUT); `
to
`pinMode(_resetPowerDownPin, INPUT);` 

and inside de if set the pin to OUTPUT

`
if (digitalRead(_resetPowerDownPin) == LOW) {	// The MFRC522 chip is in power down mode.
			**pinMode(_resetPowerDownPin, OUTPUT);**
			digitalWrite(_resetPowerDownPin, HIGH);		// Exit power down mode. This triggers a hard reset.
			// Section 8.8.2 in the datasheet says the oscillator start-up time is the start up time of the crystal + 37,74μs. Let us be generous: 50ms.
			delay(50);
			hardReset = true;
		}
`

Rotzbua, 2017-03-15:
@pedroluis5 i thinks this bug is different, because he mentioned it in version around 1.1.9/1.2.0. actually i merged the code your solution affect in 1.2.1.

Rotzbua, 2018-01-14:
Closed in favor of #355

closed 2018-01-14 by Rotzbua