miguelbalboa/rfid #267: ESP8266 Soft WDT reset on PCD_Init

Humancell, 2016-12-27:
Curious if anyone has this working with the ESP8266 lately?

I'm using the ESP-12F, and am using the Arduino IDE v1.6.5, with Arduino/ESP8266 v2.3.0, and MFRC522 v1.1.7

Everything compiles, and I've got the connections correct, but when calling PCD_Init, I get a Soft WDT reset and stack trace.  :-(

My code:

```
#define DEBUG

#include <SPI.h>
#include "MFRC522.h"

#define RST_PIN         5          // Configurable, see typical pin layout above
#define SS_PIN          4          // Configurable, see typical pin layout above

MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

void setup(void) {
	Serial.begin(9600);		// Initialize serial communications with the PC
  delay(10);
  
  #ifdef DEBUG
  /* Print our boot message and version */
  Serial.println(F(""));
  Serial.print(F("RC522 RFID Sensor - GUID: "));
  Serial.printf("%06X\n", ESP.getChipId());
  Serial.print (F("Free memory = "));
  Serial.println (ESP.getFreeHeap());
  #endif

  Serial.print(F("SPI.begin ... "));
	SPI.begin();			// Init SPI bus
  Serial.println(F("done!"));
 
  Serial.print(F("mfrc522.PCD_Init ... "));
	mfrc522.PCD_Init();		// Init MFRC522
  Serial.println(F("done!"));
 
	//mfrc522.PCD_DumpVersionToSerial();	// Show details of PCD - MFRC522 Card Reader details
	Serial.println(F("Scan PICC to see UID, type, and data blocks..."));
}

void loop() {
	// Look for new cards
	if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
}
```

When I run the output is:

```
RC522 RFID Sensor - GUID: D546E6
Free memory = 47856
SPI.begin ... done!
mfrc522.PCD_Init ... 
Soft WDT reset

ctx: cont 
sp: 3ffef300 end: 3ffef530 offset: 01b0

>>>stack>>>
3ffef4b0:  00241001 00000001 3ffee510 3ffee468  
3ffef4c0:  00000002 3ffee430 3ffee448 402022c7  
3ffef4d0:  402010ae 00000001 00000032 402010a3  
3ffef4e0:  3fffdad0 00000010 3ffee430 40202445  
3ffef4f0:  3fffdad0 3ffef554 3ffee430 402024d6  
3ffef500:  3fffdad0 3ffef554 3ffef574 40201c93  
3ffef510:  feefeffe 00000000 3ffee500 402033a8  
3ffef520:  feefeffe feefeffe 3ffee510 40100114  
<<<stack<<<
ü!ÿü!
```

I also ran on a slightly newer version of the Arduino IDE, but exact same results ...

Ideas on debugging this?

Rotzbua, 2016-12-28:
Maybe this hints help you:
- try a older esp sdk 2.1 or 2.2?
- add a delay between `SPI.begin()` and init
I tested this library a year ago with the wemos d1 mini, everything worked. Sorry that I can not help you more.
Best

Humancell, 2016-12-29:
I completely updated everything and all libraries.  I also added the delay(100); after the SPI.begin() ...

... it's working great now!

Thanks!

closed 2016-12-29 by Humancell

platdesign, 2016-12-30:
Same here ...

```
RC522 RFID Sensor - GUID: 05FFB1
Free memory = 48288
SPI.begin ... done!
mfrc522.PCD_Init ... 
Soft WDT reset

ctx: cont 
sp: 3ffef2b0 end: 3ffef4f0 offset: 01b0

>>>stack>>>
3ffef460:  00000002 3ffee3e8 3ffefc04 4020288b  
3ffef470:  00241001 00000001 3ffee4d0 3ffef540  
3ffef480:  00000002 3ffee3e8 3ffefc04 40202b23  
3ffef490:  40201c4e 00000001 00000032 40201c43  
3ffef4a0:  3fffdad0 00000010 3ffee3e8 40202cbd  
3ffef4b0:  3fffdad0 3ffee3f6 3ffee3e8 40202d4e  
3ffef4c0:  3fffdad0 3ffee3f6 3ffee414 40203e73  
3ffef4d0:  feefeffe 00000000 3ffee4c0 40201838  
3ffef4e0:  feefeffe feefeffe 3ffee4d0 40202534  
<<<stack<<<
```

- Added delay(100) after SPI.begin but no luck.
- Tried Platformio and ArduinoIDE for build and upload.
- `espressif8266` and `espressif8266_stage` as platform.
- Checked correct wiring multiple times.
- Measured wire connections.
- Tried different `RFID-RC522` board.
- Tried `NodeMCUv2` and `NodeMCUv3` (multiple boards).

Rotzbua, 2016-12-30:
Try a different SDK version, some are really buggy.

If it still fails, please add some information about your setup.

platdesign, 2016-12-30:
@Rotzbua It seams that `RST` cant be `GPIO5`. Now im using `GPIO10` and everything works :)

- Board: NodeMCUv3
- IDE: Platformio (Atom) with `espressif8266_stage` running at 160Mhz

Humancell, 2017-01-01:
Fascinating!  I have now been having intermittent problems ... but found that removing the `RST` line from `GPIO5` has it working!

I'm going to see if I can move `RST` to another pin and get better results.

emylyano3, 2017-07-08:
@Humancell What does this mean ?

> removing the RST line from GPIO5

AlbertMN, 2018-01-12:
I get this error quite a lot. I just tried to have both the RST and SS pin to be GPIO <10, and it seems to have solved the problem (for now at least). [This site](https://techtutorialsx.com/2017/04/02/esp8266-nodemcu-pin-mappings/) helped me find out which pins were what. I kept getting the `>>stack` issue when doing simple stuff like setting variables, doing `Serial.println()`.

quattroman, 2019-05-04:
I have  the same problem, 

Here is my code;

// All of the following includes are for WIFI
#include <ESP8266WiFi.h>          //ESP8266 Core WiFi Library (you most likely already have this in your sketch)
#include <WiFiManager.h>         // https://github.com/tzapu/WiFiManager

// MQTT IOT Includes
#include <PubSubClient.h>

//ultrasonic library https://github.com/Martinsos/arduino-lib-hc-sr04
#include <HCSR04.h>

// defines pins numbers
const int trigPin = D2;  //D2
const int echoPin = D1;  //D1
const int greenPin = D5; //D4
const int redPin = D4; //D5

//
UltraSonicDistanceSensor distanceSensor(trigPin, echoPin);

// defines variables
double last_dist, cur_dist;

// wifi config
// wifi auto configs using wifi manager. Just boot the device, then connect to the "ParkingSensorSetup" network
// after connecting, if you're not automatically redirected go to http://192.168.4.1 in your web browser.

// MQTT config
#define MQTT_SERV "192.168.1.111"
#define MQTT_PORT 1883
#define MQTT_NAME ""
#define MQTT_PASS ""
#define MQTT_TOPIC "stat/garagebay1/parking"

WiFiClient client;
PubSubClient pubclient(MQTT_SERV, MQTT_PORT, client);

String macToStr(const uint8_t* mac)
{
  String result;
  for (int i = 0; i < 6; ++i) {
    result += String(mac[i], 16);
    if (i < 5)
      result += ':';
  }
  return result;
}

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an Output
  pinMode(echoPin, INPUT); // Sets the echoPin as an Input
  Serial.begin(115200); // Starts the serial communication

  //Connect to WiFi
  Serial.print("\n\nConnecting Wifi... ");
  WiFiManager wifiManager;
  wifiManager.autoConnect("ParkingSensorSetup");

  Serial.println("Wifi Connected!");
  
  pinMode(greenPin, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
  pinMode(redPin, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
  digitalWrite(greenPin, LOW);
  digitalWrite(redPin, HIGH);

}

void loop() {
  MQTT_connect();
  
  cur_dist = distanceSensor.measureDistanceCm();

  Serial.print(cur_dist);
  Serial.print("cm");
  Serial.println();

  
  if(cur_dist < 140.00  && last_dist >= 140.00){
    digitalWrite(greenPin, LOW);
    digitalWrite(redPin, HIGH);
    pubclient.publish(MQTT_TOPIC, "PARKED");
    Serial.println("Publish: PARKED");
  }

  if(cur_dist >= 140.00  && last_dist < 140.00){
    digitalWrite(redPin, LOW);
    digitalWrite(greenPin, HIGH);
    pubclient.publish(MQTT_TOPIC, "VACANT");
    Serial.println("Publish: VACANT");
  }

  last_dist = cur_dist;
  
  delay(500);
}

/***************************************************
  Adafruit MQTT Library ESP8266 Example

  Must use ESP8266 Arduino from:
    https://github.com/esp8266/Arduino

  Works great with Adafruit's Huzzah ESP board & Feather
  ----> https://www.adafruit.com/product/2471
  ----> https://www.adafruit.com/products/2821

  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Tony DiCola for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

void MQTT_connect() 
{
  // Stop if already connected.
  if (pubclient.connected()) 
  {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  // Generate client name based on MAC address and last 8 bits of microsecond counter
  String clientName;
  clientName += "esp8266-";
  uint8_t mac[6];
  WiFi.macAddress(mac);
  clientName += macToStr(mac);
  clientName += "-";
  clientName += String(micros() & 0xff, 16);

  Serial.print("Connecting to ");
  Serial.print(MQTT_SERV);
  Serial.print(" as ");
  Serial.println(clientName);
  
  if (pubclient.connect((char*) clientName.c_str(), MQTT_NAME, MQTT_PASS)) {
    Serial.println("Connected to MQTT broker");
    Serial.print("Topic is: ");
    Serial.println(MQTT_TOPIC);
  }
  else {
    Serial.println("MQTT connect failed");
    Serial.println("Will reset and try again...");
    abort();
  }

  //uint8_t retries = 3;
  //while ((ret = mqtt.connect()) != 0) // connect will return 0 for connected
  //{ 
  //     Serial.println(mqtt.connectErrorString(ret));
  //     Serial.println("Retrying MQTT connection in 5 seconds...");
  //     mqtt.disconnect();
  //     delay(5000);  // wait 5 seconds
  //     retries--;
  //     if (retries == 0) 
  //     {
  //       // basically die and wait for WDT to reset me
  //       while (1);
  //     }
  // }
  Serial.println("MQTT Connected!");

  
}

And here is the error;

164.95cm
44.04cm
Publish: PARKED
Connecting to MQTT... Connecting to 192.168.1.111 as esp8266-dc:4f:22:cf:ad:6-b6
MQTT connect failed
Will reset and try again...

Abort called

>>>stack>>>

ctx: cont
sp: 3ffefce0 end: 3ffefef0 offset: 01b0
3ffefe90:  00000000 00000000 00000000 3fff0c5c  
3ffefea0:  0000001f 0000001b cf224fdc 402006ad  
3ffefeb0:  3fffdad0 00000000 3ffeea1c 402027c8  
3ffefec0:  00000000 00000000 00000001 3ffeea4c  
3ffefed0:  3fffdad0 00000000 3ffeea1c 4020a3b8  
3ffefee0:  feefeffe feefeffe 3ffe8508 40100aed  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

Thanks

Peter

Abedi98, 2021-01-13:
> I have the same problem,
>
> Here is my code;
>
> // All of the following includes are for WIFI
> #include <ESP8266WiFi.h> //ESP8266 Core WiFi Library (you most likely already have this in your sketch) [...]

Was this problem solved for you?