miguelbalboa/rfid #270: Help Please..

craxyvirux, 2017-01-07:
Greetings! can you guys make me a sketch that prints only the rfid mifare uid and personal data that wrote in the card using "Write personal data on a MIFARE PICC" .. im very new to this kind of things, please help me out for my very first arduino-rfid project in school.. thank you..

Humancell, 2017-01-07:
This is not really the place to be making these requests.  Please use the Arduino forums to ask for assistance ... this is really just for bug reports.

You should study and use the example:  https://github.com/miguelbalboa/rfid/tree/master/examples/DumpInfo

THis will dump the information for you to review.

Rotzbua, 2017-01-07:
There are tons of tutorials. Just use a big search engine.

closed 2017-01-07 by Rotzbua