miguelbalboa/rfid #272: Version not found (0xFF). Tags not recognized

cappi10, 2017-01-08:
Hello, I've seen a couple of posts about this but my answer is not in there. 

I had to solder some pins to the RFID module and then connect wires to the arduino.
The LED of the module is lighting up, but i get this warning on the serial monitor:

`Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
`

I already have redone the soldering and checked the wiring about 10 times but all seems fine.
(This was my first time soldering but it looks alright to me though.)

Anyway to find the cause of the issue?

Rotzbua, 2017-01-26:
Did you read https://github.com/miguelbalboa/rfid#troubleshooting , what did you tried?

Rotzbua, 2017-03-01:
@cappi10 Any update? Can it be closed?

closed 2017-03-21 by Rotzbua