miguelbalboa/rfid #273: Increasing ContactLessUART baudrrate to 848kbps

Dizgah, 2017-01-09:
As you know this chip support up to 848kb baud-rate while this library is worked in the default 106kb baud-rate.
When I'm trying to change baud-rate speed register(TxSpeed bits in TxModeReg register) chip does not work correctly anymore!
WBR.

JPG-Consulting, 2017-01-10:
How do you set the register? Have you taken a look to the datasheet?

Datasheet: https://www.nxp.com/documents/data_sheet/MFRC522.pdf

The TxModeReg is on page 49. By default it seems `RxCRCEn` is set to 0 but it states that this setup is only valid at 106 kBd so you will probablly have to set it up as logic 1. If you don't want to enable CRC generation during transmission it seems the reader must remain a 106 kBd.

Maybe you should also set the `RxModeReg`...

I think most of the cards work at 106 kBd by default. If you want to change the rate you will probablly have to issue a RATS (Request for Answer To Select) to check if you may change the speed and then issue a PPS (Protocol and Parameter Selection), if available, to negotiate the speed. Once the negotiation has taken place I guess then it would be a good place to change your speed.

JPG-Consulting, 2017-01-26:
I'm having a bit of trouble with this one :(

There are some more changes that need to be made... For example the `ModWidthReg` or you'll start getting timeouts even at 212 kbds. Right now my code is working for both 106 kbds and 212 kbds, but still having problems at 424 kbds and 848 kbds.

Any help with it would be appreciated