miguelbalboa/rfid #277: Multiple RFID readers- only 2 work reliable

brittio, 2017-01-19:
I'm working on a project where I need to connect a maximum of 6 RFID readers to an arduino Mega. I built it off of the ReadUidMultiReader example. The program works reliably with the "credit card" style tags, but when i use the blue keychain fobs I have issues. (i have space concerns so i need to use the blue fobs.) Only my last 2 readers are effective all the time, while my 2nd is intermittent and the 1st & 3rd are terrible. I originally thought I had solder issues with the first 3 so I moved the physical readers... Results are unchanged. I have completely unwired/rewired everything. Still the same.. I then thought maybe it was a power issue, so I split up the 3.3v legs so half would run off the arduino and the other half are on a DC/DC converter (OKI-78SR-3.3/1.5V-W36-C). Again the results are unchanged. (I also completely unplugged GND & 3.3V and it still operates the same.) In the actual code I changed the "SS_?_PINs" around to use different board pins, and saw no difference.. The only thing I think I know is that the last 2 readers always work best. example: In the line "
byte ssPins[] = {SS_1_PIN, SS_2_PIN, SS_3_PIN, SS_4_PIN, SS_5_PIN};" if I swap the pins around to "
byte ssPins[] = {SS_5_PIN, SS_4_PIN, SS_3_PIN, SS_2_PIN, SS_1_PIN};" the readers 1 & 2 will work reliably... 

Any help would be greatly appreciated..

Rotzbua, 2017-01-19:
duplicates of #191  #263
Possible solution https://github.com/miguelbalboa/rfid/issues/191#issuecomment-242631153
SPI do not work with so much devices. There will be too much noise on the connection. You need a multiplexer to separate the connections.

brittio, 2017-01-19:
I feel like there has to be another solution.  Again, if I use the large white credit card style tags it works flawlessly with 5 readers.  My only issues start when I use the smaller blue keychain fobs.  I've ordered some other small tags with hopes that they have a better antenna.

DrLou, 2017-02-21:
we've gotten several of those inexpensive fobs/cards found to be defective as shipped. Could be as simple as that?

brittio, 2017-02-21:
Thanks for your input DrLou!  I thought that at first, but no matter how I physically rearrange the readers the fobs/cards always read the same depending on the order they are in my code.  I thought the cards were reading better, but it turns out it was only Mifare Mini piccs that read well Mifare 1k piccs all read better/worse depending on the code position of the reader.  I did find code elsewhere that would turn 1 reader on until it had a valid read and then off when it went to the next reader.  This worked amazing and at good distance, but inevitably the program would freeze.  I put a cycle indicator in the program to see if it failed at the same time each time..  It did not.  Sometimes it would go 2 cycles the next 26..  It seemed to be completely random.  So after too many days I reworked my project to use specifically spaced magnetic sensors and I can not express how easy the program was to write and how flawlessly it runs compared to the rfid version.  My advice is, if you need multiple inputs use Magnetic Sensors specifically spaced and save yourself the headache of rfid readers.

closed 2017-02-21 by brittio

Annaane, 2018-06-07:
In case someone still has this problem and needs help with this issue, check this project where I made 4 RFID readers work simultaneously without any problem : https://github.com/Annaane/MultiRfid

P.S : the system works fine with both "credit card" tags and the little blue ones.

mrfarhadir, 2020-03-05:
@Annaane Hi, I followed your tutorial on youtube with two cards but I'm getting these errors:
```
16:22:44.789 -> Reader 0: Firmware Version: 0x0 = (unknown)
16:22:44.857 -> WARNING: Communication failure, is the MFRC522 properly connected?
16:22:44.926 -> Reader 1: Firmware Version: 0x0 = (unknown)
16:22:44.960 -> WARNING: Communication failure, is the MFRC522 properly connected?

```

Annaane, 2020-03-08:
@mrfarhadir Hello, I had the same issue with many readers, I just had to check the soldering is done well, I had to redo some of the pins soldering. That's all ! (and of course, make sure the wiring is correct, I have a wiring diagram in my project repo)