miguelbalboa/rfid #279: mfrc522.PICC_IsNewCardPresent() function

bob808, 2017-01-25:
Hi,
I'm trying to use this function to get an output if a card is present on the reader or not. In my application I need to keep cards on the reader, and the reader to constantly spit out the card ID (once every second or so), and at the same time a boolean marker if any card is present or not.
But I can't get it to work. Any pointers?
Thnaks.

JPG-Consulting, 2017-01-26:
I really don't understand what you are trying to do...

However, the samples use some convenience functions such as dumping the card info. Thos methods call `HaltA` so there is no need to call it again. Your code is incomplete but reading the results you obtain I guess you ar not calling `HaltA` when you get the ATQA (Which is requested in the convenience function `PICC_IsNewCardPresent` Yo could try something like this:

    void loop() {
      // Look for new cards
      if ( ! mfrc522.PICC_IsNewCardPresent()) {
        Serial.println("No card");
        return;
      }
    
      Serial.println("Present");
      mfrc522.PICC_HaltA();
    }

bob808, 2017-01-26:
I am trying to get a boolean marker that says wether a tag is present or not at all times. Thanks for the info, I'll check out your code.
What I am using right now is this:
`
void loop() {
  if ( ! mfrc522.PICC_IsNewCardPresent()) {
    delay(50);
    return;
  }
  if ( ! mfrc522.PICC_ReadCardSerial()) {
    delay(50);
    return;
  }`

It works for spitting out the tag. The code in the first post I used trying to add a marker for when any tag is present.

bob808, 2017-01-26:
I am trying to get a boolean marker that says wether a tag is present or not at all times. Thanks for the info, I'll check out your code.
What I am using right now is this:
void loop() { if ( ! mfrc522.PICC_IsNewCardPresent()) { delay(50); return; } if ( ! mfrc522.PICC_ReadCardSerial()) { delay(50); return; }

It works for spitting out the tag. The code in the first post I used trying to add a marker for when any tag is present.

bob808, 2017-01-26:
I used your code and indeed I got No card or Present on serial.
What I am trying to do, is get a Present and another action. I'm sending the code to a file on a local network server. I can't seem to make it do another thing along with Present in the loop.
I am using this code:

```
#include  "MFRC522.h"
#include "SPI.h"
#define RST_PIN 5 
#define SS_PIN  4
MFRC522 mfrc522(SS_PIN, RST_PIN);  
#include <ESP8266WiFi.h>
#include "lwip/opt.h"
void setup() {
  Serial.begin(115200);
  SPI.begin();           
  mfrc522.PCD_Init();  
  delay(10);
  }
  
void loop() {
  if ( ! mfrc522.PICC_IsNewCardPresent()) {
    Serial.println("No card");
   // delay(500);
    return;
  }
  if ( ! mfrc522.PICC_ReadCardSerial()) {
    delay(500);
    return;
  }
  dump_byte_array(mfrc522.uid.uidByte, mfrc522.uid.size);
  Serial.println("Present");
  mfrc522.PICC_HaltA();
}
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}
```
I tried it in all combinations and can't make it print the card number consistently. In the actual form, whenever I scan a card I get the card's serial number followed by "Present" but the goes to "No card" continuously even if the card is on the reader. I have to take it off and put it back again to get another serial read.

bob808, 2017-01-27:
Strangest thing. I was having trouble with your code as well. If I cycle the reader without a card in the proximity, all I get is No card, even if I get the card close to the reader.
But, if I start the reader with the card on it, it starts with "Present" and then it works as it should. I think this threw me off. I need to start again using your code.
But why would that happen? How can I make it work normally?
This is exactly the code I used:

```
#include "MFRC522.h"
#include "SPI.h"
#define RST_PIN 5 
#define SS_PIN  4
MFRC522 mfrc522(SS_PIN, RST_PIN);  
#include <ESP8266WiFi.h>
#include "lwip/opt.h"
void setup() {
  Serial.begin(115200);
  SPI.begin();           
  mfrc522.PCD_Init();  
  delay(10);
  }
  
void loop() {
  if ( ! mfrc522.PICC_IsNewCardPresent()) {
    Serial.println("No card");
    return;
    delay(50);
  }

  Serial.println("Present");
  mfrc522.PICC_HaltA();
  delay(50);
}
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}
```

simbadMarino, 2017-01-27:
Are you looking ffor a dirty way of detecting wether  card IS or IS NOT
near your RFID antenna?
If so I might have a solution, but this solution involves constantly
reseting the driver. If this helps I can provide an example :)

bob808, 2017-01-27:
I'm trying to see if there's a better way :) I need to update a file on my server constantly with the present tag code, and check if it's present or not.

bob808, 2017-01-27:
I could use a magnet and a hall effect sensor for presence detecting. Would be a cleaner way than resetting. But still messy.

simbadMarino, 2017-01-29:
Ok BOb, let us know if you find a clean way to do that, if not and still
need the dirty eway let me know ;)

bob808, 2017-01-30:
Ok mate, thanks!
So far I think I'm going to go with hall effect sensors but I'd love to have a way to only use the rfid reader.

saeedrc, 2017-02-28:
i am using this 
```

// Look for cards
 if ( ! mfrc522.PICC_IsNewCardPresent())
 {
    delay(50);
  if ( ! mfrc522.PICC_IsNewCardPresent())
  {
    delay(50);
  
    // no card detect
    return;
    }
  }

 //card detect

```

Rotzbua, 2017-03-29:
@bob808 Can this be closed?

closed 2017-04-19 by Rotzbua

imash96, 2018-03-18:
@simbadMarino @Rotzbua Can you please provide me with code that read the RFID card continuously  PLEASE PLEASE

imash96, 2018-03-18:
I am using RFID multi read code `#include <SPI.h>
#include <MFRC522.h>

constexpr uint8_t RST_PIN = D0;     // Configurable, see typical pin layout above
constexpr uint8_t SS_1_PIN = D1;   // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 1
constexpr uint8_t SS_2_PIN = D2;    // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 2
constexpr uint8_t SS_3_PIN = D3;    // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 3
String y[] = {"A0A1925E", "D0589D5E", "D0A8BEA3"};
String x[3];
constexpr uint8_t NR_OF_READERS = 3;
int state = 0;
byte ssPins[] = {SS_1_PIN, SS_2_PIN, SS_3_PIN};

MFRC522 mfrc522[NR_OF_READERS];   // Create MFRC522 instance.

/**
   Initialize.
*/
void setup() {

  Serial.begin(115200); // Initialize serial communications with the PC
  while (!Serial);    // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)

  SPI.begin();        // Init SPI bus

  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
    Serial.print(F("Reader "));
    Serial.print(reader);
    Serial.print(F(": "));
    mfrc522[reader].PCD_DumpVersionToSerial();
    mfrc522[reader].PCD_SetAntennaGain(mfrc522[reader].RxGain_max);
  }
}

/**
 * Main loop.
 */
void loop() {

  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    // Look for new cards

    if (mfrc522[reader].PICC_IsNewCardPresent() && mfrc522[reader].PICC_ReadCardSerial()) {
      Serial.print(F("Reader "));
      Serial.print(reader);
      // Show some details of the PICC (that is: the tag/card)
      Serial.print(F(": Card UID:"));
      dump_byte_array(mfrc522[reader].uid.uidByte, mfrc522[reader].uid.size);
      Serial.println();
      Serial.print(F("PICC type: "));
      MFRC522::PICC_Type piccType = mfrc522[reader].PICC_GetType(mfrc522[reader].uid.sak);
      Serial.println(mfrc522[reader].PICC_GetTypeName(piccType));

      // Halt PICC
      mfrc522[reader].PICC_HaltA();
      // Stop encryption on PCD
      mfrc522[reader].PCD_StopCrypto1();
    } //if (mfrc522[reader].PICC_IsNewC
  } //for(uint8_t reader
}

/**
 * Helper routine to dump a byte array as hex values to Serial.
 */
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}`

Den0110, 2018-06-25:
Hi. I had a similar problem and I solved it very simply.
`if (!mfrc522.PICC_IsNewCardPresent()) {
    delay(50);
    if(!mfrc522.PICC_IsNewCardPresent())
      return false;
  }
  if (!mfrc522.PICC_ReadCardSerial()) {
    delay(50);
    if(!mfrc522.PICC_ReadCardSerial())
      return false;
  }`

ElleryWa, 2019-02-15:
I realize that you might need a more complicated solution for many applications, but if you're looking for a really quick and easy way to test out continuous tag reading, here's one solution:

Just start with the ReadNUID example from the MFRC522 library, and comment out the following lines:
`  // Halt PICC
  //rfid.PICC_HaltA();

  // Stop encryption on PCD
  //rfid.PCD_StopCrypto1();`

If you're interested in running a really simple version of the ReadNUID example, with some extraneous features stripped out, I'm including the full file below. All I did was comment out parts of the original example -- no additions. I found this was a good way to check that my hardware was working correctly, without adding additional complications from the software.

`
#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN 10
#define RST_PIN 9
 
MFRC522 rfid(SS_PIN, RST_PIN); // Instance of the class

//MFRC522::MIFARE_Key key; 

// Init array that will store new NUID 
//byte nuidPICC[4];

void setup() { 
  Serial.begin(9600);
  SPI.begin(); // Init SPI bus
  rfid.PCD_Init(); // Init MFRC522 

/*  for (byte i = 0; i < 6; i++) {
    key.keyByte[i] = 0xFF;
  }

  Serial.println(F("This code scan the MIFARE Classsic NUID."));
  Serial.print(F("Using the following key:"));
  printHex(key.keyByte, MFRC522::MF_KEY_SIZE); */
} 
 
void loop() {

  // Look for new cards
  if ( ! rfid.PICC_IsNewCardPresent())
    return;

  // Verify if the NUID has been readed
  if ( ! rfid.PICC_ReadCardSerial())
    return;

  Serial.print(F("PICC type: "));
  MFRC522::PICC_Type piccType = rfid.PICC_GetType(rfid.uid.sak);
  Serial.println(rfid.PICC_GetTypeName(piccType));

  // Check is the PICC of Classic MIFARE type
  if (piccType != MFRC522::PICC_TYPE_MIFARE_MINI &&  
    piccType != MFRC522::PICC_TYPE_MIFARE_1K &&
    piccType != MFRC522::PICC_TYPE_MIFARE_4K) {
    Serial.println(F("Your tag is not of type MIFARE Classic."));
    return;
  }
/*
  if (rfid.uid.uidByte[0] != nuidPICC[0] || 
    rfid.uid.uidByte[1] != nuidPICC[1] || 
    rfid.uid.uidByte[2] != nuidPICC[2] || 
    rfid.uid.uidByte[3] != nuidPICC[3] ) {
    Serial.println(F("A new card has been detected."));

    // Store NUID into nuidPICC array
    for (byte i = 0; i < 4; i++) {
      nuidPICC[i] = rfid.uid.uidByte[i];
    }
   */
    Serial.println(F("The NUID tag is:"));
    Serial.print(F("In hex: "));
    printHex(rfid.uid.uidByte, rfid.uid.size);
    Serial.println();
    Serial.print(F("In dec: "));
    printDec(rfid.uid.uidByte, rfid.uid.size);
    Serial.println();
  //}
  //else Serial.println(F("Card read previously."));

  // Halt PICC
 // rfid.PICC_HaltA();

  // Stop encryption on PCD
  //rfid.PCD_StopCrypto1();
}

/**
 * Helper routine to dump a byte array as hex values to Serial. 
 */
void printHex(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}

/**
 * Helper routine to dump a byte array as dec values to Serial.
 */
void printDec(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], DEC);
  }
}`

roboticboyer, 2019-11-02:
My understanding is UID is read only the 1st time a card is detected using PICC_IsNewCardPresent
See #437 for more details

shnop, 2020-08-07:
Was this question ever resolved? I'm trying to get a motor to run while a the PICC is present on the sensor. From what I've read here and on other forums it looks like this is not possible without extra hardware. Is my understanding correct?

I have the same issue as @bob808 that when the card is present, the output from mfrc522.PICC_IsNewCardPresent() alternates between 1 and 0.

ans27g, 2020-09-16:
Alternating seems to be a feature of  the library or hardware
so we may have to live with it.
My suggestion.

flag=0;
  if (mfrc522.PICC_IsNewCardPresent()) 
    {
      flag=1; // card present
    }
    else  //try again to catch the toggle effect
    {
      if (mfrc522.PICC_IsNewCardPresent())
      flag=1; // card present      
    }

  if (flag)
    { 
    Serial.println("Card present"); 
    }
    else
    { 
    Serial.println("No card"); 
    }

    delay(500);

LeNouk, 2022-06-28:
Similar to @ans27g suggestion, you can simply also do something like:
```
if (!mfrc522.PICC_IsNewCardPresent() && !mfrc522.PICC_IsNewCardPresent()) { 
    Serial.println("PICC_IsNewCardPresent ---> Hell NO!");
    return;
} else {
    Serial.println("PICC_IsNewCardPresent ---> YES!");
}
```