miguelbalboa/rfid #280: Serial.print 4 readers in sequence each time a new tag is read

HiMyNamesL, 2017-01-26:
Hello, 

I currently have 4 readers wired up and working individually, however I would like to be able to Serial.print a value from the 4 readers together, but only when one of the readers reads a new tag. 0,0,0,1 for example. 

Im wanting to use 2 tags with the 4 readers and change the Serial.print depending on which tag is at which reader, 0,1,0,2 for another example. 

Thanks in advance!

HiMyNamesL, 2017-01-26:
@Rotzbua sorry to bother, and sorry for breaking rules if I had. 

I assume this isnt the place to ask for help, I saw the question tags and assumed it was.

Rotzbua, 2017-01-27:
@HiMyNamesL No problem. The `question` tag scopes (unimplemented) functions of the mfrc522. This issue is still not closed so others can still answer your question, but I think you get answers quicker from development forums. Best.

HiMyNamesL, 2017-01-27:
Awesome, thanks once again!

Rotzbua, 2017-04-19:
Closed due no answers from somebody.

closed 2017-04-19 by Rotzbua