miguelbalboa/rfid #281: A humble request to anyone who can rewrite MFRC522 library to use UART imterface

cyberjuputer, 2017-01-28:
Hi. I am not a skilled nor an experienced programmer. Even after looking at the datasheet, changing the current SPI interface to UART is totally beyond my level.

Therefore, I really hope someone will implement the MFRC522 library which uses UARt interface.

Rotzbua, 2017-01-28:
Maybe this user did it #221 
duplicate #257

cyberjuputer, 2017-01-29:
Is there any way to contact him? I am new to github and it seems to not have PM function.

Anyway, I will try to figure out what he did from his comments. There is so little information from his comments but I will give it a shot.

Dizgah, 2017-01-31:
As I remember ,Framing in SPI is different with UART in MFRC522,datasheet completely described framing for each connection.Please read datasheet and if your problem does not resolved contact me via MHDizgah@gmail.com

cyberjuputer, 2017-01-31:
Hi @Dizgah 

Yes, I have read the datasheet and I can differentiate how data is sent between SPI and UART. However, I do not know how to implement it. As I have stated in my first post, my knowledge in programming is limited. If you have some examples that I could follow through, that would be helpful. 

P/S: The problem is that the MFRC522 library uses SPI entirely. How do I modify which lines to use UART? And as far as I know, UART is a serial communication and in Arduino, there is no library for UART. You just use it. Thats why I am very confused.

omersiar, 2017-04-11:
This Library:

https://github.com/zodier/MFRC522-UART-Arduino

closed 2017-12-03 by Rotzbua

davidjonz2005, 2018-06-10:
https://github.com/davidjonz2005/MFRC522WITHUART