miguelbalboa/rfid #284: 2 new readers not working simultaneously: Firmware Version: 0x0 = (unknown)

oakmusic, 2017-02-06:
Hi!

I've been using this great library for a while with a bunch RC522 readers bought on several stores (eBay, Aliexpress, Amazon, ...) and worked fine. Now I am developing a project with 2 simultaneous readers and a Nano V3, based on ReaderMulti example, and I always get a firmware version like this when I initialize them:
**Reader 0: Firmware Version: 0x92 = v2.0
Reader 1: Firmware Version: 0x92 = v2.0**

But I recently bought 10 RC522 from this Aliexpress store (apparently the same as the old ones):
https://es.aliexpress.com/item/Free-Shipping-RFID-module-RC522-Kits-S50-13-56-Mhz-6cm-With-Tags-SPI-Write-Read/32523771442.html?spm=2114.13010608.0.0.khHkqZ

And I can not connect 2 simultaneosly. When I connect just one (alone or with one of the old) I get this firmware version on it:
**Reader 1: Firmware Version: 0x12 = (unknown)**
A bit strange, but it works

But when I connect 2 of these new readers, the firmware versions I get are:
**Reader 0: Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
Reader 1: Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?**

and no one works.

Any idea what's happening? What am I doing wrong?
The circuit is welded on a PCB and the old readers work fine in the same circuit.
Also, it seems to work when I combine an old reader with a new one (getting 0x12 firmware in the new).
I tried to power the 3.3V from a different source than the Nano 3V output, but I got the same result.

It seems that it is something wrong with the firmware, but, is there any way to flash the RC522 firmware? I couldn't find it.

Thanks for your help

Rotzbua, 2017-02-07:
Ask the seller for more information about the chip and firmware.

oakmusic, 2017-02-08:
I already asked the seller, but I had no answer. It is the same problem appeared in this forum, but nobody solved it:
https://forum.arduino.cc/index.php?topic=428898.0

Rotzbua, 2017-02-08:
What is the result with `DumpInfo`?

oakmusic, 2017-02-09:
Using a blank card, when I connect a "new reader" (alone or combine with and "old reader"), the result of DumpInfo is:

_Firmware Version: 0x12 = (unknown)
Scan PICC to see UID, SAK, type, and data blocks...
Card UID: 0E 3B E7 80
Card SAK: 08
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         62   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         61   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         60   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  14     59   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         58   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         57  MIFARE_Read() failed: Timeout in communication.
         56  MIFARE_Read() failed: Timeout in communication.
  13     55  PCD_Authenticate() failed: Timeout in communication.
  12     51  PCD_Authenticate() failed: Timeout in communication.
  11     47  PCD_Authenticate() failed: Timeout in communication.
  10     43  PCD_Authenticate() failed: Timeout in communication.
   9     39  PCD_Authenticate() failed: Timeout in communication.
   8     35  PCD_Authenticate() failed: Timeout in communication.
   7     31  PCD_Authenticate() failed: Timeout in communication.
   6     27  PCD_Authenticate() failed: Timeout in communication.
   5     23  PCD_Authenticate() failed: Timeout in communication.
   4     19  PCD_Authenticate() failed: Timeout in communication.
   3     15  PCD_Authenticate() failed: Timeout in communication.
   2     11  PCD_Authenticate() failed: Timeout in communication.
   1      7  PCD_Authenticate() failed: Timeout in communication.
   0      3  PCD_Authenticate() failed: Timeout in communication._

The same card with an "old reader":

_Firmware Version: 0x92 = v2.0
Scan PICC to see UID, SAK, type, and data blocks...
Card UID: 0E 3B E7 80
Card SAK: 08
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         62   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         61   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         60   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  14     59  PCD_Authenticate() failed: Timeout in communication.
  13     55  PCD_Authenticate() failed: Timeout in communication.
  12     51  PCD_Authenticate() failed: Timeout in communication.
  11     47  PCD_Authenticate() failed: Timeout in communication.
  10     43  PCD_Authenticate() failed: Timeout in communication.
   9     39  PCD_Authenticate() failed: Timeout in communication.
   8     35  PCD_Authenticate() failed: Timeout in communication.
   7     31  PCD_Authenticate() failed: Timeout in communication.
   6     27  PCD_Authenticate() failed: Timeout in communication.
   5     23  PCD_Authenticate() failed: Timeout in communication.
   4     19  PCD_Authenticate() failed: Timeout in communication.
   3     15  PCD_Authenticate() failed: Timeout in communication.
   2     11  PCD_Authenticate() failed: Timeout in communication.
   1      7  PCD_Authenticate() failed: Timeout in communication.
   0      3  PCD_Authenticate() failed: Timeout in communication._

but, when I connect 2 new reader, I can not read any of them:
_Reader 0: Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
Reader 1: Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?_

In my project I just get the UID, so the content of the card it's not critical for me, but whe I connect 2 of those reader I can not ready anything.

I've also tried to do pinMode(SS_PIN, OUTPUT) as said in this forum:
https://lowpowerlab.com/forum/moteino/moteino-rc522-rfid-module-uncompatible-(solved)/
but doesn't work for me

Rotzbua, 2017-02-09:
Firmware Version: 0x12  is not known and I never saw a data sheet with this. Maybe this are fake or defect chips. May have a look on the label at the chip. Additional open an issue with your seller on ali, this often helps to get information or money back. Best.

lmmeng, 2017-02-14:
Hi @oakmusic ,
I suppose the new type readers are putting a higher load on MISO.
May you take (and post) some oscilloscope screen-shots for this signal in both, working and not working configurations?

DrLou, 2017-02-21:
We've seen an 0x12 result seen on occasion - with our Firmware Version: 0x92 = v2.0 boards. _( we are using only 0x92 boards...)_

Errors have always been down to some error in wiring, often due to poor connections on a cheap breadboard.

oakmusic, 2017-02-24:
Immeng: I don't have an oscilloscope, sorry.

DrLou: That's not a wiring problem, I have tested it welded on PCB and not welded (with jumpers) on 2 different bredboards, and all the others RC522 work fine. This is just happening with these 10 RC522 lot. One works fine (with 0x12 firmware) but can not connect 2 of them simultaneously.

Anyway, thank you all for your help. Until I find a solution, I will use them in pair with a good one.

Rotzbua, 2017-02-24:
@oakmusic (1) What was the response of the seller?
(2) Maybe they just modified the version value. Try to perform a firmwarecheck by modify the method.

oakmusic, 2017-02-24:
no response from the seller :-/

Rotzbua, 2017-04-07:
@oakmusic Do you have a conclusion, so we can close this issue?

oakmusic, 2017-04-08:
I couldn't get a response from the seller. I suppose it is a hardware and / or firmware problem, which makes them incompatible with each other. I am using them in combination with other RC522 with 0x92 standar firmware and by now, they have not given problems in this way. Thanks for all your responses. You could close this issue. Thank you!

closed 2017-04-09 by Rotzbua