miguelbalboa/rfid #285: Change UID

Sn4p3, 2017-02-08:
Hi, I have a question. Either I am to stupid or the problem is not related to me but I can't find a way to change the UID of my two tags.

I first tried the ChangeUID.ino but it gave me the following errors:
`Card did not respond to 0x40 after HALT command. Are you sure it is a UID changeable one?
Error name: Timeout in communication.
Activating the UID backdoor failed.
`

Since that doesn't work i tried the ReadAndWrite.ino example project, changed the sector to 0, the blockAddr to 0 and the trailerBlock to 3. But that doesn't work too :/ 
`PCD_Authenticate() failed: Timeout in communication.`

I have no clue what I did wrong or how to do it.
Does anyone know a little bit more?

Rotzbua, 2017-02-08:
Seems you do not have a changeable PICC.

Rotzbua, 2017-03-01:
@Sn4p3 Any update?

s1102658, 2017-03-09:
Hi there,
I came across similar problems with some Chinese-made changeable UID tags and I solved the problem with the MIFARE_UnbrickUidSector(bool logErrors) function. For some reasons I don't know, sometimes changeable UID tags can be "bricked". It would be great to learn the cause of this problem and put it into good use.
By the way, I would also like to suggest a minor change to the MIFARE_UnbrickUidSector(bool logErrors) function so that end-user can pass a 16-bytes UID for unbricking and assigning at the same pass as follows.
bool MFRC522::MIFARE_UnbrickUidSector(bool logErrors, byte block0_buffer[] ) {
MIFARE_OpenUidBackdoor(logErrors);
//byte block0_buffer[] = {0x01, 0x02, 0x03, 0x04, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};

// Write modified block 0 back to card
MFRC522::StatusCode status = MIFARE_Write((byte)0, block0_buffer, (byte)16);
if (status != STATUS_OK) {
	if (logErrors) {
		Serial.print(F("MIFARE_Write() failed: "));
		Serial.println(GetStatusCodeName(status));
	}
	return false;
}
return true;
}

Cheers,
Joseph

Rotzbua, 2017-04-07:
Hi @s1102658, why should there be a bypass?

s1102658, 2017-04-08:
Hi, Rotzbua, I guess the UnbrickUidSector function is what you needed to revive a changeable UID tag. If this is the case, isn't it be more convenient to unbrick and assign a new UID in one pass? That's what I thought.

s1102658, 2017-04-10:
Hi Rotzbua,

I came across similar problems of Sn4p3 that a changeable UID tag was not
responding to UID assignment function. Then, I tried the
MIFARE_UnbrickUidSector(bool logErrors), and it successfully 'unbricked'
the tag. Then, I think it will be more convenient for end-user to unbrick
as well as assign new UID at the same time. Therefore, I would like to
suggest a slight modification of the UnbrickUidSector Function as I posted
on Mar. 9. May I know your comments.

BTW, I would like to know the ins-and-outs of the mechanism of a tag being
'bricked'.
Thanks.

Rotzbua, 2017-05-28:
@s1102658 Bypass a uid by `UnbrickUidSector` wouldnt work, because you have to calculate the `BCC` then manually.

If you want to fix that sometimes the card is bricked, maybe this more professional library helps you:
https://github.com/nfc-tools/libnfc/blob/master/examples/nfc-mfsetuid.c

closed 2017-05-28 by Rotzbua

Carlokko, 2017-08-22:
Hi there, 
I am running the rc522 on an arduino and also have the same problem as @Sn4p3.  What I dont understand is if I need to run the MIFARE_UnbrickUidSector(bool logErrors) function @s1102658 mentioned, and if so, where, because if I try to run it on the arduino IDE it returns this error.

`Archiving built core (caching) in: /var/folders/ds/wvfbdsfj1058ffv4dymbqfzm0000gp/T/arduino_cache_869140/core/core_arduino_avr_uno_51f02b7210b938436b779d1c032618e1.a
libraries/rfid-master/MFRC522Hack.cpp.o (symbol from plugin): In function "MFRC522Hack::MIFARE_OpenUidBackdoor(bool) const":
(.text+0x0): multiple definition of "MFRC522Hack::MIFARE_OpenUidBackdoor(bool) const"
sketch/FixUID.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/rfid-master/MFRC522Hack.cpp.o (symbol from plugin): In function "MFRC522Hack::MIFARE_OpenUidBackdoor(bool) const":
(.text+0x0): multiple definition of "MFRC522Hack::MIFARE_SetUid(unsigned char const*, unsigned char, bool) const"
sketch/FixUID.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/rfid-master/MFRC522Hack.cpp.o (symbol from plugin): In function "MFRC522Hack::MIFARE_OpenUidBackdoor(bool) const":
(.text+0x0): multiple definition of "MFRC522Hack::MIFARE_UnbrickUidSector(bool) const"
sketch/FixUID.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
/var/folders/ds/wvfbdsfj1058ffv4dymbqfzm0000gp/T//cc9TbvL4.ltrans0.ltrans.o: In function "main":
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:43: undefined reference to "setup"
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:46: undefined reference to "loop"
collect2: error: ld returned 1 exit status
exit status 1
Errore durante la compilazione per la scheda Arduino/Genuino Uno.
`
Thanks.

s1102658, 2017-08-23:
Hi Carlokko,

The error message is greek to me. I guess I need to know your codes before
I can debug.

Rgds,
Joseph

jonmorad, 2018-05-21:
what does it mean?
how can i solve it?
C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\user\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10801 -build-path C:\Users\user\AppData\Local\Temp\arduino_build_915818 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\Arduino\hardware\tools\avr -verbose C:\Users\user\Documents\Arduino\block2\block2.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\user\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10801 -build-path C:\Users\user\AppData\Local\Temp\arduino_build_915818 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\Arduino\hardware\tools\avr -verbose C:\Users\user\Documents\Arduino\block2\block2.ino
Using board 'mega' from platform in folder: C:\Program Files\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\user\AppData\Local\Temp\arduino_build_915818\sketch\block2.ino.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\user\AppData\Local\Temp\arduino_build_915818\sketch\block2.ino.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\user\Documents\Arduino\libraries\rfid-master\src" "C:\Users\user\AppData\Local\Temp\arduino_build_915818\sketch\block2.ino.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\user\Documents\Arduino\libraries\rfid-master\src" "C:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\user\Documents\Arduino\libraries\rfid-master\src" "C:\Users\user\Documents\Arduino\libraries\rfid-master\src\MFRC522.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\user\Documents\Arduino\libraries\rfid-master\src" "C:\Users\user\Documents\Arduino\libraries\rfid-master\src\MFRC522Debug.cpp" -o "nul"
[... cut ...]
C:\Users\user\Documents\Arduino\libraries\rfid-master\src/MFRC522.h:418:33: error: 'MIFARE_OpenUidBackdoor' was not declared in this scope

 MIFARE_OpenUidBackdoor(logErrors);

                                 ^

Using library SPI at version 1.0 in folder: C:\Program Files\Arduino\hardware\arduino\avr\libraries\SPI 
Using library rfid-master at version 1.3.6 in folder: C:\Users\user\Documents\Arduino\libraries\rfid-master 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

sebapaik, 2018-08-17:
@s1102658,
I just received my block 0 card from China, but I'm having the same issue as @Sn4p3 . I'm not sure how I can "unbrick" it. I've tried using the FixBrickedUID.ino from the examples, but the program doesn't even run. The seller also mentions that the card's default password is FFFFFFFFFFFF(F 12 times). Maybe I can use that as the key in the ChangeUID.ino?

TECH7Fox, 2018-08-26:
im also getting the same problem just like @sebapaik, i cant change the UID and FixBrickedUID.ino only shows the first rule and than it stops doing anything.

s1102658, 2018-08-30:
Hi guys,

Have your try what Rotzbua mentioned last May?
'
If you want to fix that sometimes the card is bricked, maybe this more
professional library helps you:
https://github.com/nfc-tools/libnfc/blob/master/examples/nfc-mfsetuid.c
'
As far as I know, the codes of miguelbalboa had been updated and improved
last year.  You had better consult directly to Rotzbua.
BTW, I came across some Chinese vendors showing wordings of changeable UID
card/tag on their web sites but were actually not true.  Sometimes it takes
time and effort to source a good one.

Cheers,
JC

jjwer210184, 2019-08-09:
@s1102658 thanks for solution you provided, but wonder if there is a way to fix this problem without using c programing.

lee-jongwoo, 2023-01-06:
If your card doesn't respond to the magic backdoor commands it might be a gen2 card, which is supposed to be written directly without using a backdoor. In this case, try calling `mfrc522Hack.MIFARE_SetUid(newUid, (byte)4, key, false)` from [the v2 library](https://github.com/OSSLibraries/Arduino_MFRC522v2). The last parameter tells it to try writing without using the backdoor, which will probably work with next-gen CUID cards.

nightshadow931, 2023-03-19:
> If your card doesn't respond to the magic backdoor commands it might be a gen2 card, which is supposed to be written directly without using a backdoor. In this case, try calling [...]

Just wanted to say this actually works with gen2 cards, you only need to change line
  if ( mfrc522Hack.MIFARE_SetUid(newUid, (byte)4, key, **true**) ) {
to
  if ( mfrc522Hack.MIFARE_SetUid(newUid, (byte)4, key, **false**) ) {