miguelbalboa/rfid #287: RFID Arduino activated content

abryan156, 2017-02-11:
Greetings,
I'm very new to Arduino and am trying to get my RFID Arduino set up to open content. Either a webpage or a file on my computer. Any ideas are greatly appreciated.
Thanks

bghost4, 2017-02-11:
You will have to write a program that sits on the computer, and is
monitoring the serial Port, then when you scan an rfid card it will send
data over to the computer program that will decide what the computer will
do.

abryan156, 2017-02-11:
Thank-You, that is very helpful. Do you have any tutorial suggestions that might help me write this kind of program? I'm really new to all of this, any help you could offer would be really appreciated.

bghost4, 2017-02-11:
Hmm I can't think off any direct tutorials, but in java you could use
awt.desktop class to open a file or webpage. There is also a serial Port
library... Jserialport or something like that. It would allow you to
connect to the arson serial Port and look at incoming data.

bghost4, 2017-02-11:
The serial Port library is jssc.

abryan156, 2017-02-11:
Thanks so much for your help, this is a huge lead.

Rotzbua, 2017-02-12:
@abryan156 Can it be closed?

abryan156, 2017-02-12:
Sure!

closed 2017-02-12 by Rotzbua