miguelbalboa/rfid #288: ESP8266 Resets

homebrew79, 2017-02-16:
I am trying to use an RC522 on an Amica NodeMCU board, with the ESP-12E. My sketch seems to run great for a while and then the board will randomly reset. To troubleshoot the issue I started with a very minimal sketch, let it run for a while, added a little more, and kept going until the resets started happening again.

Here is the sketch with the minimum it takes to cause the resets. It runs fine until I add the #include "MFRC522.h" line.

```
#include "Arduino.h"
#include "SPI.h"
#include "MFRC522.h"

const int RST_PIN = 5; // RST-PIN
const int SS_PIN  = 4;  // SDA-PIN

void setup() {
    Serial.begin(115200);    // Initialize serial communications
    delay(250);
    Serial.println(F(""));
    Serial.println(F("Booting...."));

    SPI.begin();

    Serial.println(F("Ready!"));
    Serial.println(F("======================================================"));
}

void loop() {
}

```
 The RFID board is not hooked up. I believe my power to be good. It is a dedicated power supply. And I don't think it's the power because it runs fine without that line.

Here is the serial output when it resets.

```
Booting....
Ready!
======================================================

Exception (0):
epc1=0x402013de epc2=0x00000000 epc3=0x00000000 excvaddr=0x4020137f depc=0x00000000

ctx: cont
sp: 3ffef110 end: 3ffef2d0 offset: 01a0

>>>stack>>>
3ffef2b0:  3fffdad0 00000000 3ffee2a0 402013f6
3ffef2c0:  feefeffe feefeffe 3ffee2b0 402020f0
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe8430
~ld
```

Thanks for your help.
Please let me know if I can provide any more information.

-Brian

Rotzbua, 2017-02-17:
Did you checkout #267 ?
Otherwise add some information about your software setup. IDE version, esp SDK version, ..
Sry, can't help you more. Best.

homebrew79, 2017-02-17:
Thanks Rotzbua!

I did see #267 before, but my situation is a little different because it doesn't happen right on PCD_Init. It usually runs for several minutes to hours before resetting.

I am using platformio: IDE 1.7.2 | Core 3.2.1

```
Platform Manager
================
Platform espressif8266
--------
Updating espressif8266 @ 1.3.0:         [Up-to-date]
Updating tool-mkspiffs @ 1.102.0:       [Up-to-date]
Updating tool-esptool @ 1.409.0:        [Up-to-date]
Updating tool-espotapy @ 1.0.0:         [Up-to-date]
Updating framework-arduinoespressif8266 @ 1.20300.1:    [Up-to-date]
Updating toolchain-xtensa @ 1.40802.0:  [Up-to-date]
```

```
Library Dependency Graph

|-- <MFRC522> v1.2.0
|   |-- <SPI> v1.0
|-- <SPI> v1.0
```

I hope this information is helpful.

lmmeng, 2017-02-18:
Hi,
ESP8266 needs time for "internal jobs". Maybe you are not offering enough?
See https://github.com/esp8266/Arduino/blob/master/doc/reference.md#timing-and-delays

homebrew79, 2017-02-21:
Thanks for the suggestions. It seems to have been a problem with the board I was using. After switching boards, everything is stable, and my program has been running for over 24 hours without any resets.

I'm sorry for any time wasted on this.

closed 2017-02-21 by homebrew79