miguelbalboa/rfid #290: Up to 5 devices simultanious

Lennyz1988, 2017-02-22:
I am posting this to describe my experience in trying to get 5 RFID devices (MFRC522) to work at the same time using 1 Arduino.

This is also relevant to these posts:

https://github.com/miguelbalboa/rfid/issues/191

https://github.com/miguelbalboa/rfid/issues/277

https://github.com/miguelbalboa/rfid/issues/263

After a lot of trial and error I got 5 devices to work simultaneously using this library. 

- Make sure to use quality cables! I used CAT5E cables to connect the RFID devices to the Arduino.

- I started with 5 cables with a total length of 40 meters. I could get 2 devices to work reliable but I could not use the other 3 devices. So I repositioned the Arduino and cut the cables to about 15 meters length total. Maybe if you would use even better quality cables you can increase the length of the cables. 

- Try multiple RFID cards. I noticed after a while that some cards would work on one device but were not recognized on another device. 

- Make sure to have 1 or 2 backup MFRC522. When I couldn't figure out why one device would not read the tags when connected, I swapped it with one of my backups and it started working. Keep in mind that the device that could not read the card still works when connected on my breadbord. So maybe there is just to much interference for that device. 

So now after many hours of trial and error I got 5 devices to work reliably, without any extra components, over a total length of 15 meters.

I hope this helps someone.

masterleo, 2017-04-11:
Hello ! We have the same problem it's realy annoying ! 
How did you manage to get 5 devices working ? can you show us the code in witch 5 devices work (what I have never seen) !!

masterleo, 2017-04-11:
Help me on this issue here : 
http://arduino.stackexchange.com/questions/36982/multiple-mfrc522-rfid-reader-on-uno-bugs

Lennyz1988, 2017-04-11:
You can just use the standard ReadUidMultiReader example.

- Add the additional ss pins in the code
- add the ss pins to the "byte sspins"

That's it.

The code you posted makes no sense at all to me. I suggest you start from scratch.

MDLSoft, 2017-06-07:
Hello, I'm working with 4 readers and it works ok, It's based on miguelbalboa's multi rfid example
`bool getRFID(byte readern)
{
  bool isPICCpresent = false;
  mfrc522[readern].PCD_Init();   // Init the reader
  if (mfrc522[readern].PICC_IsNewCardPresent() && mfrc522[readern].PICC_ReadCardSerial())
    {
     memcpy(readedCard[readern], mfrc522[readern].uid.uidByte, 4); // Copy UID to a global array
     isPICCpresent = true;
    }
  mfrc522[readern].PICC_HaltA();
  mfrc522[readern].PCD_StopCrypto1();
  digitalWrite(RST_PIN, LOW);    // return to hard low power mode
  return isPICCpresent;          // returns TRUE if PICC is detected, false if not
}`

But I have a problem setting the gain, I tryed this after the PCD_init():

mfrc522[readern].PCD_SetAntennaGain(0x07);

and:

mfrc522[readern].PCD_SetRegisterBitMask(mfrc522[readern].RFCfgReg, (0x07<<4));

But when I try to change the gain, my readers can't read the rfid tags or read it sometimes. It is annoying that i can't setting the gain fine.

When I did a basic program to test the mfrc522 readers, setting the gain worked fine, but now with the multi-readers and the need of a loop with the PCD_Init() (I need to know that the card is present yet), it doesn't work!

I would appreciate some help. Thanks!

akeilox, 2017-06-08:
i think i saw a snipped of this in a forum but it was deleted when i check back.
can you post the entire .ino file so i can check with similar setup?

MDLSoft, 2017-06-08:
Here is the .ino, this is a test setup, I have a more complex program comparing UIDs, storing it to eeprom, activating outputs and many others functions but the base is this, reading the UIDs of the PICC. I don't need access to PICC memory or do more advanced stuff, only read their UID and know when PICC is over the reader. You can see in the sketch where I tryed to set gain.

```C++
//******************************************************************************************
//**                     ARDUINO PRO MINI 3.3v 8MHz                                       **
//**                    GND  GND  VCC  RX   TX   /DTR                                     **
//**                 +--------------------------------+                                   **
//**                 |  [ ]  [ ]  [ ]  [ ]  [ ]  [ ]  |                                   **
//**                 |          SERIAL PORT           |                                   **
//**             D1  | [ ]1/TX                 RAW[ ] |   |                               **
//**             D0  | [ ]0/RX                 GND[ ] | <--> GND                          **
//**                 | [ ]RST        SCL/A5[ ] RST[ ] |                                   **
//**                 | [ ]GND        SDA/A4[ ] VCC[ ] |  --> VCC 3.3v                     **
//**             D2  | [ ]2/INT0    ___         A3[ ] |  --> SDA/SS reader #3             **
//**             D3  |~[ ]3/INT1   /   \        A2[ ] |  --> SDA/SS reader #2             **
//**             D4  | [ ]4       /PRO  \       A1[ ] |  --> SDA/SS reader #1             **
//**             D5  |~[ ]5       \ MINI/       A0[ ] |  --> SDA/SS reader #0             **
//**             D6  |~[ ]6        \___/    SCK/13[ ] |  --> SCK (Clock)                  **
//**             D7  | [ ]7          A7[ ] MISO/12[ ] |  <-- MISO (Master In Slave Out)   **
//**             B0  | [ ]8          A6[ ] MOSI/11[ ]~|  --> MOSI (Master Out Slave In)   **
//**             B1  |~[ ]9                  SS/10[ ]~|  --> RST All readers (for now)    **
//**                 |           [RST-BTN]            |                                   **
//**                 +--------------------------------+                                   **
//**              http://busyducks.com/ascii-art-arduinos                                 **
//**                                                                                      **
//**               MDLSoft(c) 2017   Test setup ver: 1.02                                 **
//******************************************************************************************

#include <SPI.h>
#include <MFRC522.h>
#include <string.h>

constexpr uint8_t NR_OF_READERS = 4;

constexpr uint8_t RST_PIN  = 10;
constexpr uint8_t SS_0_PIN = A0;
constexpr uint8_t SS_1_PIN = A1;
constexpr uint8_t SS_2_PIN = A2;
constexpr uint8_t SS_3_PIN = A3;

byte ssPins[] = {SS_0_PIN, SS_1_PIN, SS_2_PIN, SS_3_PIN};

byte readedCard[NR_OF_READERS][4];  // Matrix for storing UID over each reader
MFRC522 mfrc522[NR_OF_READERS];     // Create MFRC522 instances

void setup()
{
  Serial.begin(9600); // Initialize serial communications
  while (!Serial);    // Do nothing until serial connection is opened
  SPI.begin();        // Init SPI bus
  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) 
   {
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
   }
 pinMode(RST_PIN, OUTPUT);
 digitalWrite(RST_PIN, LOW);    // mfrc522 readers hard power down.
}

void loop() 
{
 for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) 
  {
    if(getRFID(reader))
    {
      Serial.print(F("Reader "));
      Serial.print(reader);
      Serial.print(F(": Card UID:"));
      printUID(readedCard[reader]);
      Serial.println();
    }
  }
delay(2000);
}

//******************** Routine for print 4 byte UID to serial ******************************
void printUID(byte *buffer){
for (byte i = 0; i < 4; i++){
Serial.print(buffer[i] < 0x10 ? " 0" : " ");
Serial.print(buffer[i], HEX);}}
//********************************END OF ROUTINE********************************************

//************ Routine for scan readers and store the UIDs in readedCard array *************
bool getRFID(byte readern)
{
  bool isPICCpresent = false;
  digitalWrite(RST_PIN, HIGH);   // Get RC522 reader out of hard low power mode
  mfrc522[readern].PCD_Init();   // Init the reader
  //****** Trying to set antenna gain to max, erratic functioning of the readers ***********
  //mfrc522[readern].PCD_SetRegisterBitMask(mfrc522[readern].RFCfgReg, (0x07<<4)); 
  //mfrc522[readern].PCD_SetAntennaGain(0x04);
  // mfrc522[readern].PCD_ClearRegisterBitMask(mfrc522[readern].RFCfgReg, (0x07<<4));
  // mfrc522[readern].PCD_SetRegisterBitMask(mfrc522[readern].RFCfgReg, 0x07);
  //delay(50); 
  if (mfrc522[readern].PICC_IsNewCardPresent() && mfrc522[readern].PICC_ReadCardSerial())
    {
     memcpy(readedCard[readern], mfrc522[readern].uid.uidByte, 4); 
     isPICCpresent = true;
    }
  mfrc522[readern].PICC_HaltA();
  mfrc522[readern].PCD_StopCrypto1();
  digitalWrite(RST_PIN, LOW);    // return to hard low power mode
  return isPICCpresent;          // returns TRUE if PICC is detected, false if not
}
//********************************END OF ROUTINE********************************************
```

Khamphai, 2017-06-11:
I use  Arduino UNO Board and i have a problem with two RFID by pin MOSI and MISO, i follow the code on top my comment, so can help me

MDLSoft, 2017-06-13:
You shoul use 3.3v for the RFID readers, you could lower the UNO from 5v to 3.3v or use level shifters. Other than that is to have the connections right and as is a relatively high frequency you should take care about contacts in the breadboards or dupont wires. In my case, I soldered flat cables to have better conectivity.

You could show us your circuit, maybe we can help you with it.

roooney, 2017-06-14:
Hi there, I am looking to connect 6 RC522  RFID readers and I have wired them according to this image but I was wondering, do I have to also connect the 3.3V from the Arduino to each of the RC522 Readers as well?  Thank you for your help!
[image]

MDLSoft, 2017-06-14:
Yes, you have to use 3.3v to Vcc for every RC522. All the communication bus (MOSI, MISO and SCK) and select (SSn) should also work at 3.3v. 

Maybe you have already though about this, but I suggest you try first with only one Rc522 reader and then add one by one. It's easier to find bugs and get working multiple rfid readers in the same bus. I haven't tried more than 4 readers but you shouldn't have any problems.

roooney, 2017-06-14:
Thank you MDLSoft, so i'll dasiy chain all RC522's 3.3V like I would for other wires (MOSI, MISO, etc). Thank you!

remconet, 2018-01-30:
I was hoping I´d find what I´m looking for but sadly no... What I want it is also to use 5 readers, but like access control. Every rfid reader is programmed to ´allow´ 1 tag/card. So when all 5 tags are scanned by the proper reader, a realy switches. How do I do that? It's like combining acces control with multirfidreader but I don't know how to do it.

DrLou, 2018-01-30:
Wouldn’t be too hard to do what you’re proposing…  Of course the readers themselves are not actually ‘programmed’ to remember cards. This must be handled by your logic. One approach would be to store your ‘valid keys’ in the EEPROM of your MCU device…

Opening a relay thereafter is trivial. There are good demo sketches around for both of these elements.

Lou

amkamyab, 2018-04-20:
Hello Everyone
i have a question. is it important that all RFID reader have same Firmware?
i want use 5 RFID ( RC522 ) with one arduino.
when i join one RFID it work well and read cards. when i make it 2 RFID it work well yet and rad cards.
but when i join 3 RFID, it don't work and have Error. 
my RFIDs version is:
Reader 0: Firmware Version: 0x92 = v2.0
Reader 1: Firmware Version: 0x12 = (unknown)
please help me with that

Thanks

Annaane, 2018-04-27:
I also managed to make 5 RFID readers to work simultaneously with Arduino Uno and miguelbalboa's library. Check this repo https://github.com/Annaane/MultiRfid or this is a video that shows that's functional : https://www.youtube.com/watch?v=ahc8Yai_sWI

thoka, 2018-05-09:
@amkamyab  I had the same issue before I initialized all CS pins to HIGH before the individual setup. Otherwise all RFID devices will respond simultaneously and you get crazy firmware readings.

pexes, 2018-05-19:
Hello, I also faces this problem and i found library similar to Sunfounders. But there is no SPI thing so you can't initialize many readers, however, you can rewrite it. It is ny following link:
https://www.bitsandparts.eu/documentation/77/Programmeer_voorbeeld.ino

amkamyab, 2018-06-30:
@thoka 
hello.
may i ask you to explain more what i should do to fix this problem.
and something else. how long distance can have my RFID cable.
Thanks
Amkamyab

sjoerd1999, 2018-07-25:
I also had this problem. 
I first had a project where I used 3 RFID readers and it worked fine.
Later I had to rebuild my project, with 3 RFID readers from other manufacturers, and it didn't work.
The new readers worked individually, but not combined.

I solved this by turning the readers | on, measure, off | one by one. This worked really well.
The readers need about 20mA, which the digital pins can provide.
So what you can do is: 
digitalWrite(1,HIGH);
receiveData();
digitalWrite(1,LOW);
for all readers.

!! BUT !!
The readers need 3.3v, and the digital pins provide 5v. So I used a 3.3v regulator to convert the 5v to 3.3v. Otherwise the RFID readers might explode. I used the LD1117V33.

You can switch them really quickly and it still works. (I tried 10 readings per second).

I anyone needs more information on this, I'll gladly help out! Just message me

suimani, 2018-08-22:
Dear Github Swarm, 
I've been looking into your reign whenever I needed help and found it without needing to ask. 
The time has come to approach you directly. 

This is going to be a long one, just trying to explain the situation as good as possible.

First off, please don't be too harsh, my coding skills are almost none existent, and it takes me a really long time to understand what's happening in a code. While making progress, I'm happily spending more hours than necessary. 
The following code I have (partly) written myself and I am almost proud that it is indeed working (again partly).
It is an escape room puzzle (or at least should be, once it works)

The Situation: 
An Arduino nano has 3 RFID readers attached. 
Each of those 3 RFID readers only shall read their specific RFID tag correctly. So for each reader, there is only one correct tag. 
We have 3 lights for each of those readers. A reader reads the correct tag - the corresponding light will shine.
Once all 3 readers read their correct tags at the same time (and all 3 lights are shining) a Relais is activated to power(or depower) a 12V magnet for a small door. There is a timer involved, which works fine, once the readers work. 

The Problem: 
When plugged in via USB it works perfectly fine. 

When plugged in via 12V adapter it works as well, but ONLY every second or third time. 
Sometimes it works a couple of times in a row, then a couple of times not. I can't find a logic in the pattern (if one can call it a pattern)

One last fun fact: in the code below Reader#2 is initiated last, - this is the one that doesn't always work. 
It is ALWAYS the last initiated reader which doesn't work. 

Why does it work when powered via USB, and then not always when powered via 12V VIN?

tags, Cables, Readers, Nanos have been replaced to make sure it's not a hardware error. 

To come to an end- I appreciate all help I can get, 
so thank you in advance, 
best regards, 
Mani

```
#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN 10
#define RST_PIN 9
#define SS_PIN2 8
#define SS_PIN3 7
#define LED_R 2 //define red LED
#define Light1 3
#define Light2 4
#define Light3 5
#define relais 6

MFRC522 mfrc1(SS_PIN, RST_PIN);   // Create mfrc1 instance.
MFRC522 mfrc2(SS_PIN2, RST_PIN);
MFRC522 mfrc3(SS_PIN3, RST_PIN);

int zustand1;
int zustand2;
int zustand3;

bool timerGestartet = false;
unsigned long timeAfter2Seconds = 0;

unsigned long currentTime = 0;

void setup()
{
  Serial.begin(9600);   // Initiate a serial communication

  SPI.begin();      // Initiate  SPI bus

  
  
  mfrc1.PCD_Init();
[... 170 lines cut ...]
    Serial.print(mfrc3.uid.uidByte[i] < 0x10 ? " 0" : " ");
    Serial.print(mfrc3.uid.uidByte[i], HEX);
    content.concat(String(mfrc3.uid.uidByte[i] < 0x10 ? " 0" : " "));
    content.concat(String(mfrc3.uid.uidByte[i], HEX));
  }
  Serial.println();
  Serial.print("Message : ");
  content.toUpperCase();
  if (content.substring(1) == "B0 A2 89 A7") //change here the UID of the card/cards that you want to give access
  {
    Serial.println("Authorized access");
    Serial.println();
    digitalWrite(Light3, HIGH);
   
  }

  else   {
    Serial.println(" Access denied");
    Serial.println();
    digitalWrite(LED_R, HIGH);
    delay(1000);
    digitalWrite(LED_R, LOW);
  }
}
```

sjoerd1999, 2018-08-22:
Maybe the 12V isn't clean DC ? Like it isn't producing a constant 12V, but rather jumps back and forth around 12V. This could cause problems.
I run MRFC522 RFID cards with a 9V battery all the time, and it works fine.

Other than that I don't know a reason why it doesn't work. Maybe the supply doesn't provide enough current?
You can also try to use a 5V regulator to turn the 12V into 5V and run that into the 5V pin on the Arduino.
You can also just use a powerbank to power it, but I understand it would be better to use the 12V because you also need it for the lock.

I did excaclty the same thing as you. I made an escape room puzzel with 3 RFID chips.
If the right 3 tags are present, a 4-digit code is shown on a small LCD. I had to use a 3.3V regulators and write complicated code, but that was because of multiple chips not working even on USB. Your chips seem to be working fine though so you don't need that.

sjoerd1999, 2018-08-22:
`/**
   -----------------------------------------------------------------------------------------
               MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
               Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
   Signal      Pin          Pin           Pin       Pin        Pin              Pin
   -----------------------------------------------------------------------------------------
   RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
   SPI SS 1    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required *
   SPI SS 2    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required *
   SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
   SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
   SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15

*/

#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN         9          // Configurable, see typical pin layout above

#define SS_1_PIN        6         // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 2 /3
#define SS_2_PIN        7          // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 1 / 3
#define SS_3_PIN        8          // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 1 / 2

#define NR_OF_READERS   3         // How many readers you have?

byte ssPins[] = {SS_1_PIN, SS_2_PIN, SS_3_PIN};

MFRC522 mfrc522[NR_OF_READERS];   // Create MFRC522 instance.

String currentIDs[] = {"00000000", "00000000", "00000000"}; // Current ID's, changes as you play
String correctIDs[] = {"53806d1d", "1692a012", "738811d"}; // Which ID's are correct? , stays constant as it checks this with the current ID's

void setup() {
  Serial.begin(9600); // Initialize serial communications with the PC
  SPI.begin();        // Init SPI bus
}

void loop() {
  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    currentIDs[reader] = "00000000";
    digitalWrite(reader + 2, HIGH);
    delay(40);
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
    delay(40);
    if (mfrc522[reader].PICC_IsNewCardPresent() && mfrc522[reader].PICC_ReadCardSerial()) {
      currentIDs[reader] = getCode(mfrc522[reader].uid.uidByte, mfrc522[reader].uid.size);
      mfrc522[reader].PICC_HaltA();
      mfrc522[reader].PCD_StopCrypto1();
    }
    digitalWrite(reader + 2, LOW);
  }

  // Print out the current ID's
  Serial.print(currentIDs[0]);
  Serial.print(" , ");
  Serial.print(currentIDs[1]);
  Serial.print(" , ");
  Serial.println(currentIDs[2]);

  if (currentIDs[0] == correctIDs[0] && currentIDs[1] == correctIDs[1] && currentIDs[2] == correctIDs[2]) { // If all ID's are correct
    Serial.println("ALL 3 CORRECT, YOU WIN");
  }

}

String getCode(byte *buffer, byte bufferSize) {
  String code = "";
  for (byte i = 0; i < bufferSize; i++) code = code + String(buffer[i], HEX);
  return (code);
}`

haaslukas, 2018-09-12:
Has anyone ever tried using multiple MRFC522 RFID readers bought from china (I guess they are clones)? They work perfectly when using one only, but it looks like there is some noise on the MOSI line even if I turn off the readers that I do not want to read by actively pulling the corresponding SS pins high. The ReadUidMultiReader code works, when I disconnect all the MOSI cables exept from one, but of course that is just helpful for debugging. Or am I missing something else?

sjoerd1999, 2018-09-12:
> Has anyone ever tried using multiple MRFC522 RFID readers bought from china (I guess they are clones)? They work perfectly when using one only, but it looks like there is some noise on the MOSI line [...]

Hi, please look at my posts above.
I got chips from AliExpress that had this problem.

The only solution I found is:
Turn chip 1 on,
Read data chip 1,
Turn chip 1 off.
And repeating that for all sensors.

You are supposed to turn the chips on/off not with the SS pin, but with the 3.3V power input of the MFRC522. The readers need 3.3v, and the digital pins provide 5v. So I used a 3.3v regulator to convert the 5v to 3.3v. Otherwise the RFID readers might explode. I used the LD1117V33.

Please take a look at the code above.

After hours and hours of trying this is the best method I could find.

You can email me at sjoerd1999@gmail.com in case you still have questions or want more info.

MDLSoft, 2018-09-16:
Of course, using RST pin you use the same SS pin for all readers. This is as I have configured. I suppose that cut the supply line or using the RST is the same, in both cases requires same initialization time. How did you that? The supply is directly one pin from arduino?
For me using the RST pins was the most obvious and easy, one arduino pin to each RST, and shared SS pin. In my case each arduino's output haven't to support any current than internal pull-up but since only one reader working at time, no problem for the arduino.
I desoldered all leds to reduce the consumption, not a drastic change but sure better durability of batteries. 

I haven't any problem using 5v input modules supplying it with only 3.3v, or maybe only a simple transistor as digital switch when "power" inductance connected without optocoupler isolation.

sjoerd1999, 2018-09-16:
The Arduino digital pins can provide 20mAh, the same as the chips take. So yeah the supply is directly from the Arduino pin in my case.
What code do you use to do the RST thing?
digitalWrite(RSTpin, LOW) when sensing, digitalWrite(RSTpin, HIGH) when off?

MDLSoft, 2018-09-16:
Yes, the way as I use RST is as you said:
digitalWrite(RSTpin, LOW) when sensing
digitalWrite(RSTpin, HIGH) when off

I think our sketches should very similar because the concept of working is about the same.

sjoerd1999, 2018-09-16:
The RST pin can take 5V, so then a 3.3V regulator wouldn't be necessary. I'm def gonna try this out. Thank you for sharing your findings!

frederic-bonjour, 2018-09-24:
> Since there's only one reader on at the same time, all chips could also use the same SS pin to save some pins. Don't know if that would work with the RST pin method though.
> 
> Using a 3.3V Arduino is smart! Only a bit annoying if you want to use the Arduino to run a 5V component as well.
> 
> Could you explain why you are using the RST pin to power it off? In my method I just cut the 3.3V line. Is the RST method better?
> 
> Normally the readers take 20mAh each, so when you're only running them one by one for a short period of time, this saves a huge amount of power indeed!

Hi,
Thank you for the tip of turning on and off the readers, to only have one at a time! Clever!
Did you have that working during a long time?
Do the readers like being turned on and off (quite quickly)?
Do you have a schematics of your installation, that works with the code you posted above?
Thank you for your answers!

haaslukas, 2018-09-28:
I've found another solution without turning off the devices multiple times per second: I've inserted an Logic Buffer (CD74HC125E) between the different MISO lines and are turning these lines off via the SS pins. No re-programming needed. The ReadUidMultiReader demo sketch works perfectly with three cheap AliExpress modules and an ESP8266 (WeMos D1) board.

vogonsaudi, 2019-04-08:
Note - I've been using the method MDLSoft posted for the last year or so on multiple projects.
I did have a problem using an ESP32 where the wires were touching, but the more I think about it, the more I think it may have been a power issue.

Anyway - I have found that the version of the MFRC522 library is important. For the ESP8266, I had to upgrade to the latest version 1.4.1 for it to work, but that upgrade caused this method to fail for standard arduino Unos. So, I downgraded to 1.3.5 and it worked.

MDLSoft, 2019-04-22:
Of course, MFRC522 version is important and it depends the functions used in your sketch, I have used deprecated functions and I didn't spent time to upgrade my sketch to the new version of the library, for me is working fine with older version. So to anybody who wants to use my sketch or similar ones with deprecated functions should take in care that the library version is important to work correctly.

I have to say that the system has been working perfectly for two years without any failures with cheap aliexpress modules, but I think is much mor interesting use a bit more expensive PN523 or another i2c RFID reader easier to connect and programming.

vogonsaudi, 2019-05-05:
So, at this point, would you recommend using your method posted above or switch to the model that's included in the original library's example file? (where you run a common SDA and individual RST pins)?

I'm actually stuck on a project and I guess may bad RC522 cards, I'm not sure, but I'm thinking about switching to the other method just to try it.

MDLSoft, 2019-05-09:
I really don't know what to say you. If you have all allready cabled my method with the correct library version sould work. 

You should think about the time and effort you need to do this RC522 cards to work , and if it worth your  time. 

I did it with a lot of time and effort, too many wires for RC522, connections and possibility of failures, so I spent many money with good cabling, connectors etc, for sure I should leave RC522 cards to PN523 or other like this, a bit more expensive but much more comfortable to develop.

If you want to use this RC522 cards and my method doesn't work for you (a bit extrange) you should try the original library methods and use the newer one.

PavelasV, 2019-11-04:
Hello, I just finished a "weekend" fight with 5 Chinese RC522 sensors, so short story: I created a prototype with 5 great [vma405 chips from velleman ](https://www.velleman.eu/products/view/?id=435568) MFRC522.h lib recognize them as 2.0v chips and they are working great(i connected them in parallel, also using logic shifters and additional 3.3v power supply unit), I am also using ethernet shield on the same SPI bus, so wrote the code, created the PCB, ordered it, and because of desire to save some bucks ordered rc522 Chinese version - which was twice cheaper... And they are really shitty... Chinese chips are messing all SPI bus if you connect more than one reader to SPI, no device will work on that bus(one by one - they are working OK). So was looking why this is happening, and found  that they overloading MISO line and of course making some noise on it, so the decision was to put simple diode 1n4847 on each reader MISO pin directed to MCU pin(rc522 -->| MCU), and one resistor(1.5k ohm) from MCU MISO to GND and voila, as diode isolated one reader  MISO  from another reader MISO, issue fixed! hope that would be helpful for somebody. 
[image]

ThierryR83, 2020-02-12:
Hello,

Can i use the 1N4148 for replace the 1N4847 ?

Merci.

MDLSoft, 2020-02-13:
> Hello,
> 
> Can i use a diode 1N4148 for replace the 1N4847 ?
> 
> Merci.

Yes.
I Think maybe PavelasV typed wrong the diode number, it's a bit strange to use a 48v Zener here.

ThierryR83, 2020-02-13:
Ok,
Thank you for your reply.

sarfer00, 2020-07-05:
Hi @giz02 , you said you've got 12 RFID working with one Arduino. I have 12 RFID with one arduino and I've tested sharing MISO line, sharing SS, but I can't get it work.
I have read this post several times and you and others talk about pullup and pullout de readers, but I don't understand how. Could you explain to me how I should connect the wires?
@MDLSoft , @sjoerd1999 , you also talk about pullup and pullout the readers through RST line. Could you help me?
Thank you very much.

sjoerd1999, 2020-07-05:
> Hi @giz02 , you said you've got 12 RFID working with one Arduino. I have 12 RFID with one arduino and I've tested sharing MISO line, sharing SS, but I can't get it work.
> I have read this post [...]

In my original method, I wired all sensors to the same Arduino pins, except the power(3.3V) pins, which each get a unique digital pin. I then only had one sensor on at the same time by only giving power to one sensor, this way they don't interfere. This works great, only the downside is that the reader needs 3.3V, while most Arduino's are 5V. So you need to step down this 5V from the digital pin to 3.3V for each sensor. I did this using a 3.3V regulator for each sensor (LD33CV), this could also be done with a MOSFET/transistor potentially. (if you have a 3.3V Arduino you don't need the regulators/MOSFETS)

But then MDLSoftproposed that you can do the same idea, only have one sensor on at once, by instead of cutting power to the sensor, just switching the reader on/off by using the RST pin of the sensor. This way you don't need the 3.3V regulators or MOSFETS. The wiring is really straightforward, just connect all sensors to the same Arduino pins, except the RST pins, which each get a unique digital pin.

I've attached code below.
RFIDmulti is my previous method with power pins (tested, works with library version 1.3.6)
RFIDmultiRST is MDLSoft's method with RST pins (not tested)

[RFIDmulti.zip](https://github.com/miguelbalboa/rfid/files/4875467/RFIDmulti.zip)

Rumble25, 2020-07-14:
I am having a hard time using the RFIDmulti with reset. It reads the first one and then spits out empty codes 000000 for the rest of the readers

Rumble25, 2020-07-14:
Thanks for quick response...I am trying these at the moment. I am struggling to read any cards at the moment will update

Rumble25, 2020-07-14:
At the moment with those updates I have still struggled to get any reading on the code. Any other thoughts?

sjoerd1999, 2020-07-14:
> At the moment with those updates I have still struggled to get any reading on the code. Any other thoughts?

Just build the setup really quick with two readers.
My RST code works, only change I needed to make is swap the HIGH and LOW in the digitalwrite() when turning on/off the reset pins :)

Rumble25, 2020-07-14:
got it thanks I guess i changed something else by accident. Genuinely appreciate the time have a good one

sarfer00, 2020-08-09:
> In my original method, I wired all sensors to the same Arduino pins, except the power(3.3V) pins, which each get a unique digital pin. I then only had one sensor on at the same time by only giving [...]

Hi @sjoerd1999 . Thank you very much for your quick response. I've tested your code but  I don't get it works. I've bought too LD33V regulators but it doesn't works neither. The sensors are activated one after another like expected but they don't read the UID. 
I have checked the voltage received in the sensors and it is less than 3.3V (2,54V approx), i dont know if its because the output of the arduino are not 5v but between 3,23V and 3,64V.

I am using: Arduino Mega 2560, RFID sensord RC522 and LD33V regulators.

What I am doing wrong? 
Thank you

sjoerd1999, 2020-08-09:
> Hi @sjoerd1999 . Thank you very much for your quick response. I've tested your code but I don't get it works. I've bought too LD33V regulators but it doesn't works neither. The sensors are activated [...]

Using an older library version might help (I used the MFRC522 library from the Arduino IDE library manager, version 1.3.6).

When using the LD33V method, do the lights of the RFID readers turn off completely when not reading, or are they slightly dimmed? I had a problem with some versions of the RC522 where the chips did not turn off completely when disconnected from power and therefore still interfered. They should all blink one-by-one.
Also make sure you connect the LD33V properly, (GND, Arduino digital pin, RC522 3.3V pin)

The RST method worked fine for me, maybe check the wiring again? Also make sure you use the right Arduino code, both methods have different code.

:)

sarfer00, 2020-08-09:
I am downloading the library version 1.3.6 for testing... if not works i will test the RST method again....
Yes, the readers turn off completely when not reading, they blink one by one. And the LD33V is connected properly ... yes, Ive tested it with external 5V and the output is 3.3V. Input Voltage Min for LD33V is 4.4V and Iam checking that is not enough voltage at input, this is the most strangest thing...

sarfer00, 2020-08-09:
Library 1.3.6 doesn't resolve the problem. Somebody know why digital output gives 3,6V instead 5V ??

sjoerd1999, 2020-08-09:
> Library 1.3.6 doesn't resolve the problem. Somebody know why digital output gives 3,6V instead 5V ??

How are you powering the Mega? If you power it via the Vin pin, you need at least 7V.
If the digital pins only give 3.6V, you can also omit the LD33V, as the RC522 can handle voltages up to 3.6V. But it'll be better to fix the 5V problem to prevent stability issues...

sarfer00, 2020-08-09:
I am powering trough USB for monitoring, and after I will power it with the electric transformer. 
If I connect only a wire the digital output is 4,4.
If I connect the LD33V with the output to RFID sensor the digital output is 3,32 (and only 2,36 reaches the RFID sensor). 
If I connect the LD33V with the input to RFID sensor (same wire from digital output to RFID 3,3v pin through LD33V) the digital output is 3,03 (and then only 3,03 reaches the RFID sensor). 
I am breaking my mind with this, I don't understand it....

sarfer00, 2020-08-15:
Many thanks @sjoerd1999  for all your help. I've got it works perfectly with 12 sensors!! Finally with **RST method** (that doesn't work first time I tried out)....

CiscoGD, 2020-08-15:
> Many thanks @sjoerd1999  for all your help. I've got it works perfectly with 12 sensors!! Finally with **RST method** (that doesn't work first time I tried out)....

Gents, I am facing same issues you all described here.
I need 5 RFID working although not simultaneously.
I have tried to RST method without success.
Joerd1999, do you mind share your code and how did you wire them?
I need to finalise my school project. many thanks.

sjoerd1999, 2020-08-15:
> Gents, I am facing same issues you all described here.
> I need 5 RFID working although not simultaneously.
> I have tried to RST method without success.
> Joerd1999, do you mind share your code and how did you wire them?
> I need to finalise my school project. many thanks.

This is the code [RFIDmultiRST (1).zip](https://github.com/miguelbalboa/rfid/files/5078670/RFIDmultiRST.1.zip)
Did you get one RFID reader working? What board are you using?
The wiring is quite straightforward, all readers connect to the same SDA/MOSI/MISO/SCK/GND/3.3V pins. Only the RST pin of each reader goes to a unique digital pin. These RST pins are declared in the code (the RSTpins[] array)

CiscoGD, 2020-08-16:
Thanks for such quick replay. Sorry disturbing you again:
I am using a MEGA 2560 and 5 x RFID-RC522
I´ve been testing with your code, just modifying SDA pin ( I ´v used pin 30 )  and adding digital pins up to 5  for each RST instead of 3
MOSI,MISO,SCLK and SDA  are common  connexion and from there to 50,51 and 52  and pin 30 for SDA
When running I am just getting “00000000” all the time ,  no one RFID RC522 is reading.
3.3V are supplied by same Mega, perhaps I need and external P.Supply?

[cid:image003.jpg@01D673B3.A295B9F0]

Enviado el: sábado, 15 de agosto de 2020 15:06
Para: miguelbalboa/rfid <rfid@noreply.github.com>
CC: Godall, Cisco <Cisco_Godall@instron.com>; Comment <comment@noreply.github.com>

sjoerd1999, 2020-08-16:
> Thanks for such quick replay. Sorry disturbing you again: I am using a MEGA 2560 and 5 x RFID-RC522 I´ve been testing with your code, just modifying SDA pin ( I ´v used pin 30 ) and adding digital [...]

The 3.3V pin of the MEGA can only supply up to 50 mA of current total. Since the readers need 20mA each, you might want to use a different 3.3V power supply. (A 3.3V regulator should work fine). (this might not be needed as the inactive readers don't consume as much, but still worth a try).
Please try it first with 1 or 2 readers, then scale up to 5.
Also, which digital pins are you using as RST pins? Don't use pins 0 and 1, as they are used for Serial.
You could also try to use the ICSP header pins for the MISO/MOSI/SCK pins, instead of 51/50/52

CiscoGD, 2020-08-29:
Thanks so much for your help.
After many test here below my findings.
When I use a RFID in one arruino reliability is 100 %
When 2 are connected to a single arduino reliability is 90%.
With 3 RFID reliability falls down to 40%
And with 4 MiSO together is about 20%, with 5 is no more than 10%.
As I need a very reliable system and no much time left for more experiments I have added  UNO’s for each RFID and all them connected with I2C.
Now my project is 100% reliable.
Is a very nice felling to know that there is a community ready to help.
Thanks so much.

Emès des d’iPhone

HibaHorya, 2021-08-04:
after around million of trying and trying I discovered that the correct way to connect 4 RFID RC522 is to put them in the same line on test board except SS pins and the code as usual is ReadUidMultiReader from RFID library ... like this :
[image]

nimbuldev, 2021-08-15:
Wanted to leave a comment to extend my gratitude to @sjoerd1999, his RST method and code base raised my reliability to 100% (so far) on a mega using 5 RFIDs. I did not have to account for voltage levels, all RFIDs share all ports in a daisy chain except for RST. 

Previously on each power up there was about a 40% chance one of the rfids would never scan. With this, finally, I have a robust RFID base for escape room puzzles. 

The code did not work until I downgraded my MFRC522 library to 1.3.6.

taufiqulhadi, 2021-12-22:
Hello, I have sketched two RFIDs and two load cells. When I combined the program between 2 RFIDs and 2 loadcells, the two RFIDs were read and the first loadcell was read but the second loadcell was not read. If the two RFID programs and loadcell programs are separated then the two RFIDs run normally and the two loadcells run normally. How can I combine the two programs so that they can run normally? Thank you
[Two RFID Code .txt](https://github.com/miguelbalboa/rfid/files/7761021/Two.RFID.Code.txt)
[Two Strain Gauge Code.txt](https://github.com/miguelbalboa/rfid/files/7761022/Two.Strain.Gauge.Code.txt)

[9 more replies]