miguelbalboa/rfid #292: UINT8_MAX error

lbortel, 2017-03-14:
I' ve found out compilation problem issue:

...\Arduino\libraries\rfid-master\src\MFRC522.cpp:201:28: error: 'UINT8_MAX' was not declared in this scope

Probably it has something to do with last release, because older release of library works fine.
I tried to include <stdint.h> to mfrc.cpp but still does not work. :/

Arduino IDE version: 1.7.11

Rotzbua, 2017-03-14:
In 1.6.13 it compiles also travis, but my IDE says that it couldn't be resolved.. strange

I think we also should review and remove some of this functions:
```c++
MFRC522();
MFRC522(byte resetPowerDownPin);
MFRC522(byte chipSelectPin, byte resetPowerDownPin);

void PCD_Init();
void PCD_Init(byte resetPowerDownPin);
void PCD_Init(byte chipSelectPin, byte resetPowerDownPin);
```

Rotzbua, 2017-03-14:
Answer is easy, if we would know avr-libs :/

http://www.nongnu.org/avr-libc/user-manual/group__avr__stdint.html

> C++ implementations should define these macros only when __STDC_LIMIT_MACROS is defined before <stdint.h> is included 

Does this fix your problem?

lbortel, 2017-03-16:
Solved. Thanks a lot!

TomTom0815, 2017-03-20:
I still can't compile since this check-in from end Feb'17 https://github.com/miguelbalboa/rfid/commit/637e81cabae273425f1329ea515cd201265a3946 (removed need for reset pin (#275)) anymore. This commit introduced the UINT8_MAX the first time.

@tkoester 
@Rotzbua 

    rfid/src/MFRC522.cpp
    lib/rfid/src/MFRC522.cpp: In constructor ‘MFRC522::MFRC522()’:
    lib/rfid/src/MFRC522.cpp:16:33: error: ‘UINT8_MAX’ was not declared in this scope
     MFRC522::MFRC522(): MFRC522(SS, UINT8_MAX) { // SS is defined in pins_arduino.h, UINT8_MAX means there is no connection from Arduino to MFRC522's reset and power down input
                                 ^
    lib/rfid/src/MFRC522.cpp: In member function ‘void MFRC522::PCD_Init()’:
    lib/rfid/src/MFRC522.cpp:201:28: error: ‘UINT8_MAX’ was not declared in this scope
      if (_resetPowerDownPin != UINT8_MAX) {
                            ^
    make: *** [.build/nano328/rfid/src/MFRC522.o] Error 1
    .build/nano328/Makefile:55: recipe for target '.build/nano328/rfid/src/MFRC522.o' failed

Rotzbua, 2017-03-20:
Hi @TomTom0815 ,
Please provide some information about your setup like IDE version and micro-controller target.

As mentioned in https://github.com/miguelbalboa/rfid/issues/292#issuecomment-286593758 try to add before `#include <stdint.h>`:
```c++
#define __STDC_LIMIT_MACROS
```
If it works, please make a pull request, so I can merge the fix.

TomTom0815, 2017-03-20:
Hi @Rotzbua 
Thanks for the quick reply. I'm using arduino in CLI mode only. I checked the changes with UINT8_MAX one more time on different arduino versions and with different build tools.
With the latest toolchain (1.8.1) the issue is indeed gone and I can't reproduce it anymore.
The arduino and ino (for CLI building) I used have been fairly outdated - okok, they have been freaky old (*shame on me*)

Now, I'm off to rework my CLI arduino continuous integration system to verify sketches with the latest arduino version ...

closed 2017-03-26 by Rotzbua

mmogadem, 2017-05-11:
Hi,
I have the same problem and I tried hard to use every suggested solution but I couldn't work it out!
even I reinstalled IDE 1.6.5 version instead of IDE 1.8 and also didn't work out!!
Any recommendations would be appreciated

lbortel, 2017-05-11:
Hi @mmogadem,
do you have `#define __STDC_LIMIT_MACROS` line in MFRC522.h? If not, try to add this line.

mmogadem, 2017-05-11:
Hi @lbortel 
yes, I have this line. My problem lies in this >MFRC522.cpp:201: error: 'UINT8_MAX' was not declared in this scope if (_resetPowerDownPin != UINT8_MAX) {

mmogadem, 2017-05-11:
@lbortel 
I tried using another codes and I had this message
DumpInfo:48: error: 'class MFRC522' has no member named 'PCD_Init'
DumpInfo:49: error: 'class MFRC522' has no member named 'PCD_DumpVersionToSerial'
DumpInfo.ino: In function 'void loop()':
DumpInfo:55: error: 'class MFRC522' has no member named 'PICC_IsNewCardPresent'
DumpInfo:60: error: 'class MFRC522' has no member named 'PICC_ReadCardSerial'
DumpInfo:65: error: 'class MFRC522' has no member named 'PICC_DumpToSerial'
DumpInfo:65: error: 'class MFRC522' has no member named 'uid'
'class MFRC522' has no member named 'PCD_Init'

Rotzbua, 2017-05-11:
@mmogadem No `c++11` support, no compile. IDE 1.6.5 do not support `c++11`, see patch https://github.com/arduino/Arduino/commit/ddf4d878fdb39173b872f69149fb4cf1cd9fd017

kayace, 2017-05-12:
I removed my Arduino 1.6 and I  installed Arduino 1.8.  It worked.

mmogadem, 2017-05-23:
@lbortel @Rotzbua @kayace 
Thanks all, it works!!

commit dc25caf7 (2017-03-26): fix #292

fix missing UINT8_MAX