miguelbalboa/rfid #293: WDT reset on ESP8266 (fixed)

pedroluis5, 2017-03-15:
hello 

I'm using Wemos D1 board base on ESP8266 and there's a problem with reset pin.
To solve the problem i've made some change on PCD_init function.

I fixed the problem doing this...

change the line 203 from
`pinMode(_resetPowerDownPin, OUTPUT);`
to
`pinMode(_resetPowerDownPin, INPUT);`

and inside the next if set the pin to OUTPUT

```
if (digitalRead(_resetPowerDownPin) == LOW) {	
    pinMode(_resetPowerDownPin, OUTPUT);
    digitalWrite(_resetPowerDownPin, HIGH);	
			
    delay(50);
   hardReset = true;
}
```

also do not use GPIO15(D8) for reset use another I'm using

```
#define RST_PIN D3
#define SS_PIN  D0
MFRC522 mfrc522(SS_PIN, RST_PIN);   // Create MFRC522 instance
```

Rotzbua, 2017-03-15:
@pedroluis5 thanks for your report. the affected library version is missing. (that's the reason i added the issue template ;) )

please make a pull request with your fix. also please add your mentioned problem with pin to the readme. thanks

pedroluis5, 2017-03-15:
sorry...

is the last one version 1.3.2

TonyVera, 2017-06-02:
Hi @pedroluis5 ,
Thank you for finding the problem.  I've been hitting a brick wall for two solid days trying to get my Wemos D1 mini working with several RC522 boards.   I thought your fix was going to solve my issue but it didn't.  I've verified my connections countless times.  Tried 3 different Wemos boards and 2 RC522 boards.  All results are identical:
---------
Serial Monitor Output:
chksum �**Version=Firmware Version: 0x0 = (unknown)**
**WARNING: Communication failure, is the MFRC522 properly connected?
Version=0**
Scan a MIFARE Classic PICC to demonstrate read and write.
Using key (for A and B): FF FF FF FF FF FF
BEWARE: Data will be written to the PICC, in sector #1
----------------
My Wemos D1 mini pin assignments/connections:
SS: D2
SCK: D5
MOSI: D7
MISO: D6
RST: D3
----------------
Both RC522 boards work fine on Arduino Uno (with different pin assignments of course).  Just can't get it to work on any Wemos D1 mini.

Any suggestions you could offer would be GREATLY appreciated.  Thanks!

pedroluis5, 2017-06-02:
Hello.
Good to know that I can help you with something.

I’d attached my code for a access control. I hope that can help you. Also at same zip file has the library that I’m using.

If you need something more let me know

you can send me a whatzapp if you want to.

Pedro Nascimento
55 12 99188-5322

TonyVera, 2017-06-03:
@pedroluis5  sorry, I don't see the attachment(s).  I'm new to the community so I'm probably just overlooking something.  How can I view/download an attachment?  Thanks!

pedroluis5, 2017-06-03:
Just now I realize that I answered this via email but all data come to here without the attachments. Can you pass me an email. So that I can send my code. Tks

TonyVera, 2017-06-03:
Thank you so much for being so generous with your time.  You can reply to this email with the code and library.

Have a  great weekend!

Tony

pedroluis5, 2017-06-03:
I can't see an email kkkk. Please send me a whatsapp message with your email. 

Pedro Luis do Nascimento 

+55 012 99188-5322

TonyVera, 2017-06-03:
Hi Pedro,

I don’t have WhatsApp, but I just installed it and it requires a phone # to send you a message.  I tried the number below your signature but it requires your country (which I don’t know).  Sorry I’m not familiar with WhatsApp yet.  My email address is tonyrvera@gmail.com <mailto:tonyrvera@gmail.com>.  

Thanks,
Tony

TonyVera, 2017-06-05:
@Pedro.  I tried the library you sent and it worked on my Wemos D1 Mini flawlessly.  Thank you!  Where did this library come from?  How will the fixes/improvements in this library find their way back into the main library baseline (branch)?

pedroluis5, 2017-06-05:
This library I've downloaded from here. Version 1.3.2. I just change the way of the reset pin is used. I don't know how to upload this to a branch or even the trunk base. 

I glad in help you.

TonyVera, 2017-06-05:
Hmmm... I guest there must be some sort of regression between version 1.3.2 and the latest (1.3.6).

Rotzbua, 2017-06-06:
@TonyVera No changes in pcd_init between 1.3.2 and 1.3.6
@pedroluis5 i recommend github desktop client. really simple to use. then please make a pull request. thanks :)

pedroluis5, 2017-06-06:
OK. I already use a git client. I will do that. 

Pedro Luis do Nascimento 

+55 012 99188-5322

TonyVera, 2017-06-06:
@Rotzbua Yes, agreed.  There are quite a few other changes, **outside** of PCD_Init between 1.3.2 and 1.3.6.  I don't have a clue which change fixed the problem with ESP8266 (Wemos D1 mini).  I can only report that the library sent to me by @pedroluis5 (in Arduino library manager reported as v1.3.1) works perfectly and the latest 1.3.6 fails per my first post on this thread.

5chufti, 2017-08-29:
Hi,
I'm using the arduino 1.8.1 IDE portable on W7/32 with manual integration of esp8266 support (update_sdk_2.1.0 branch from github) and rfid (master from github) and "DumpInfo" example compiles and works without any problem on NodeMCU V1.0

wiring the MFRC522 to ESP8266 (NodeMcu)
RST    = GPIO5 ( D1)
SDA   = GPIO4  (D2)
MOSI = GPIO13 (D7)
MISO = GPIO12 (D6)
SCK    = GPIO14 (D5)
GND   = GND
3.3V   = 3.3V

closed 2018-01-14 by Rotzbua

SviFi, 2019-02-09:
Maybe this helps someone. I was having a similar issue of WDT reset while trying to get **Wemos D1 mini** to work with another RFID chip (smaller, orange color, square shape **PN532**, "NFC MODULE V3" chip ). It would not upload a new sketch (gives connection errors) and also would freeze after restarting chip. The only way I could get it to work is to disconnect the PN532 chip while restarting the chip and after initialization connects it back - it would work. 

I solved it by changing SS pin from **D8** (which is otherwise recommended, including in schematics of Wemos D1 mini under SPI connection) to **D0**, as @pedroluis5 suggested here and guess what - it worked! The firmware started to upload fina and no more freezes. Thank you @pedroluis5 ! 🥇

commit feda7b55 (2018-01-14): fix #293 #324 #347, hang of PCD_Init() on ESP8266 boards (#354)

On the Wemos D1 mini ESP8266 board the RST pin was LOW all the time and the MFRC522::PCD_Init() didn't reset the MFRC522 because of a non working digitalWrite, resulting in a endless loop.
commit ff4f51f3 (2018-10-04): fix #293 #324 #347, hang of PCD_Init() on ESP8266 boards (#354)

On the Wemos D1 mini ESP8266 board the RST pin was LOW all the time and the MFRC522::PCD_Init() didn't reset the MFRC522 because of a non working digitalWrite, resulting in a endless loop.