miguelbalboa/rfid #294: library for esp8266 nonos sdk

mmmmar, 2017-03-20:
I port it to esp8266 partly.
https://github.com/mmmmar/esp8266-mfrc522

could I realase it in my repository?

Rotzbua, 2017-03-20:
Yes, why not? :)

Some points:
- It would be nice, if you can add an additional notice in your file https://github.com/mmmmar/esp8266-mfrc522/blob/master/mfrc522/mfrc522.c to this library.
- And maybe you publish your port under a licence, so other people can use your library.
- You can add your port under https://github.com/miguelbalboa/rfid#compatible-boards if you make an pull request.

Best

mmmmar, 2017-03-20:
I'm improving it @Rotzbua

Rotzbua, 2017-04-07:
Thanks for your work. I added you as reference into the readme 9e48c308da4660c29ec84eacfad5e4146c8b3a02 .

closed 2017-04-07 by Rotzbua