miguelbalboa/rfid #299: Firmware Version: 0x0 = (unknown) WARNING: Communication failure, is the MFRC522 properly connected?

javier237, 2017-03-26:
### Step 1: Are you in the right place?

### Step 2: Describe your environment

 * OS version: ___7__
 * Arduino IDE version: ___1.8.2__
 * MFRC522 Library version: __1.3.2___
 * Arduino device: _Mega____
 * MFRC522 device: ___RC522__
  
### Step 3: Describe the problem

Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
[image]

I have double checked the connections and all are done according to the typical connections of MEGA.

I have tried and replaced arduino MEGA and the RFID RC-522 but I have got no success.

Would be great if you could help.
Thank you

Rotzbua, 2017-03-26:
You didnt config the code:

```c++
#define RST_PIN         9          // Configurable, see typical pin layout above
#define SS_PIN 10 // Configurable, see typical pin layout above

javier237, 2017-03-26:
like that 
[image]

Rotzbua, 2017-03-27:
You have to change the values of the pins to that value you have connected them.
E.g.:
```c++
#define RST_PIN       5          // Configurable, see typical pin layout above
#define SS_PIN 53 // Configurable, see typical pin layout above

closed 2017-03-28 by Rotzbua

surabhi96, 2017-12-03:
@Rotzbua Hey, I am using the uno only but I still get the same error.
I have connected RST to digital pin 9 and SDA to pin 10

Rotzbua, 2017-12-03:
@surabhi96 Please use the large [troubleshooting](https://github.com/miguelbalboa/rfid/blob/master/README.rst#troubleshooting). If nothing works contact your supplier for help or replacement.

rafabrasail, 2018-01-12:
I'had the same problem. I've solved this problem welding the wires.

TheMatth, 2018-02-22:
Hey ! I had the same problem as you all, and I fixed it by connecting the wires directly from the Arduino UNO to the RFID reader without a breadboard. 
Hope this tip can help some of you ;)

Lollywash, 2018-06-02:
I was having the exact same problem,
I fixed by:
Verifying what board I have (Arduino Uno, Arduino Mega, Arduino Nano, Arduino Pro micro)
Then depending on that physically changing my pins for the correct board.
You can see all the different pins in an example file from the MFRC522 examples (I used the DumpInfo Example) 
[image]
Then in the code you must change 2 values:
"RST_PIN"
"SS_PIN"
Right here:
[image]
Hope this helps thanks

usermigallon, 2018-06-12:
Hello,
I change this for arduino mega:
// Create MFRC522 instance.
constexpr uint8_t RST_PIN = 5;     // Configurable, see typical pin layout above
constexpr uint8_t SS_PIN = 53;     // Configurable, see typical pin layout above
and works

Lollywash, 2018-06-12:
Yes that would be the correct configuration fire the Arduino Mega as stated by the example program (highlighted in my previous post)

usermigallon, 2018-06-13:
Right, thank you for your post.
Can I ask you? if i lost my tag how can deleted from code ?

anistark, 2018-09-04:
Same issue still facing. I'm using RFID RC522 and followed the exact configurations. 

```
Firmware Version: 0x0 = (unknown)
WARNING: Communication failure, is the MFRC522 properly connected?
MFRC522 software version = 0
Scan PICC to see UID, SAK, type, and data blocks...
```

Anybody found any solution?

Cevalik, 2019-03-02:
i have arduino uno and getting firmware error 0x0 (unknown)
connections are fine but don't know why it is showing communication failure

marlonmilesmg, 2019-03-04:
i have arduino uno, 2 * mfrc522 rfid readers and getting firmware error 0x0 (unknown)
connections are fine but don't know why it is showing communication failure
[image]

marlonmilesmg, 2019-03-05:
Soldering is the way to go guys. Worked for me!!!

DarkiStar, 2019-05-26:
see my PR
`https://github.com/miguelbalboa/rfid/pull/476`

mimya01, 2020-02-01:
**same problem.**

mimya01, 2020-02-01:
Soldering is the way  for me

GianPieers, 2020-09-30:
¿Y para nodemcu-spe32s funciona?

awebwemu, 2020-10-13:
> ### Step 1: Are you in the right place?
> ### Step 2: Describe your environment
>
>     * OS version: _**7**
>
>     * Arduino IDE version: _**1.8.2**
>
>     * MFRC522 Library version: **1.3.2**_
>
>     * [...]

Can use arduino uno?

tonyMath97, 2020-11-22:
Estou com o mesmo problema no Arduino  Uno, as configurações das portas estão corretas, não posso soldar, alguém alguma outra alternativa?

batmacumba, 2020-11-25:
Infelizmente tem que soldar mesmo, tony. Boa sorte!

tonyMath97, 2020-11-25:
> Infelizmente tem que soldar mesmo, tony. Boa sorte!

Como se tratava apenas de um projeto da faculdade (e não possuo solda), apenas forcei o contato do modulo RFID com a Barra de pinos que estava conectada na protoboard e funcionou perfeitamente. Mas a solução é soldar mesmo.

pyyan11ss, 2022-11-29:
tinggal solder aja si tony goblok

Rizky-anugerah, 2022-12-18:
> tinggal solder aja si tony goblok

weeeeyyyy, sans bray

AYODEJI006, 2023-05-06:
I also faced the same problem.
The firmware version error does not matter.
what i did to solve the problem is:
I use separate voltage 3.3V to power RFID reader instead of connecting it to arduino 3.3V pin
but remember to connect arduino GND to the separate 3.3V GND.
[image]

yuvrajtayal1202, 2023-05-18:
dwwcw

kyle-yeh, 2023-10-24:
It may be that the voltage is too low due to poor contact. I pressed it to make the board more in contact with it.

empflow, 2023-12-01:
> It may be that the voltage is too low due to poor contact. I pressed it to make the board more in contact with it.

This was exactly my issue! Thanks! I just had to tape it a bit better

cyberajk, 2024-02-20:
Check to make sure SS pin is not being used by another device or shield.  My shield was using SS pin 10, and when I switched the RFID board to a free pin and updated the code, it fixed the connection error.

baoOyster, 2024-04-03:
Even I've keep it hardly, nothing happen :://

kazuwsx, 2025-01-13:
@marlonmilesmg 

> Soldering is the way to go guys. Worked for me!!!

I saved this comment.
Really appreciate your comment.

Guys, soldering is the trueth.

Yar00myr, 2025-05-18:
If you could tell me what the problem is, I would be very thankful:

Arduino IDE version: _2.3.6
MFRC522 Library version: 1.14.12
Device: Freenove v4 (v4.1)
MFRC522 device: RFID-RC522

Code:
`#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN 4
#define SS_PIN 5

MFRC522 mfrc522(SS_PIN, RST_PIN);

void setup() {
  Serial.begin(9600);
  SPI.begin();
  mfrc522.PCD_Init();
  Serial.println(F("Тестування RC522..."));
  mfrc522.PCD_DumpVersionToSerial();
}

void loop() {
  if (!mfrc522.PICC_IsNewCardPresent()) return;
  if (!mfrc522.PICC_ReadCardSerial()) return;
  
  Serial.print(F("UID:"));
  for (byte i = 0; i < mfrc522.uid.size; i++) {
    Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
    Serial.print(mfrc522.uid.uidByte[i], HEX);
  }
  Serial.println();
  mfrc522.PICC_HaltA();
}`

Photo: 

[image]