miguelbalboa/rfid #302: UID 4/7 byte

kuslu, 2017-03-31:
### Step 1: Are you in the right place?
is it a bug?

### Step 2: Describe your environment

 * OS version: Windows 8
 * Arduino IDE version: 1.8.1
 * MFRC522 Library version: 1.3.2
 * Arduino device: UNO R3
 * MFRC522 device: Firmware Version: 0x92 = v2.0
  
### Step 3: Describe the problem
When i upload Dumpinfo example (also other examples show same thing) and try to read a card, it gives me 4 byte UID. But in serial monitor i see 7 byte UID on 0 block.

Ex. Output (on serial monitor):
Card UID: 45 C5 25 83
Card SAK: 08
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
           62   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
...
...
...
            0   45 C5 25 83  13 88 04 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 

#### Affected file(s) or example(s):
Tested Files:
     1- DumpInfo
     2-ReadNUID
     3-RFID-Cloner

#### Steps to reproduce:

  1. Upload example to Arduino Uno r3
  2. Put card on RFID reader (RC522)
  3. Check Serial Monitor
  3a. Look On Card UID: XX XX XX XX **(4 BYTE)**
  3b. Look Block 0 => XX XX XX XX  XX XX XX 00  00 00 00 00  00 00 00 00  [ 0 0 0 ]  **(7 BYTE)**
  
#### Observed Results:

  * What happened?  Shows us 4 Byte of UID instead of 7 Byte
  
#### Expected Results:

  * What did you expect to happen?
There were 7 Byte UID

Rotzbua, 2017-03-31:
Confirmed, but I have no time to fix. Somebody else has to fix and create a pull request.

abhifoo, 2017-04-04:
According to Page 8 and 9 of MIFARE Classic 1K Datasheet (http://www.mouser.com/ds/2/302/MF1S503x-89574.pdf), the first Data Block (block 0) of first Sector (sector 0) contains the IC manufacturer data, of which first 4 bytes (bytes 0, 1, 2 and 3) represent the NUID, which is the Manufacturer programmed 4 byte 'Non-Unique Identifier'. See the images attached. Not much information is provided on the remaining 12 bytes of that block. I think it is safe to ignore them.

[image]

[image]

Rotzbua, 2017-04-04:
@abhifoo Thanks.
In fact the old system says: UID has only 4 bytes for Mifare Classics, 7 bytes have MIFARE Ultralight and MIFARE Ultralight C.

It seems that this has be changed and there exists 4 and 7 byte UID Mifare cards.
See: http://www.scnf.org.uk/smartstore/4-7_B_ID_Questions_Answeres_V8.pdf

Any suggestions for a solution?

abhifoo, 2017-04-05:
@Rotzbua , thanks for the info.
How about ignoring bytes 4-6 if they are 00 and report a 4 Byte NUID, otherwise report the 7 Byte UID? Another solution would be to report 7 Byte UID even if bytes 4-6 are 00 and follow this format everywhere.

conoror, 2017-05-28:
If I may comment, this doesn't appear to me to be a bug.

PICC_DumpToSerial starts with PICC_DumpDetailsToSerial which outputs the UID as received when the PICC transitioned to ACTIVE from READY. If it was going to send a 7 byte UID it would have done so in two cycles: 4 bytes starting with the Cascade Tag 0x88 (which you don't use so that's really 3 bytes, the CT telling you "there's more") and 4 bytes in the second cycle which makes 7. The fact that the output is:
```
Card UID: 45 C5 25 83
```
means that there is no bug. If there was, the first byte would be "88". The UID is correct as the card transmitted it. If there is more data to be gained by dumping card data itself, well that's a separate thing. Isn't it?

saeed-parsaee, 2017-09-16:
@conoror words are completely correct.
FYI: The detail step for card with 4/7/10 bytes UID length described [here](https://www.nxp.com/docs/en/application-note/AN10927.pdf). 
[image]

luna-lovecraft, 2021-03-16:
quiero usar los demas ejemplos pero tengo error en el ejemplo de escritura de datos personales, no encuentro la opcion o parte del codigo para meter la informacion de la persona, se queda ciclado o en pausa despues de esta parte, soy nuevo en esto 
[image]
[image]