miguelbalboa/rfid #309: usage

mmmmar, 2017-04-14:
when I use your library in esp8266, it can work, but if it run a long time, it can't find card.

if I rest(soft rest) mfrc522 repeadly every 30 sec, it work well. 

do you have any idea about this?
Thanks.

Rotzbua, 2017-04-18:
Please add information for people who want to reproduce your issue:
- How long is a "long time" ?
- Do not recognize tags, but does the mfrc522 react in general? (e.g. firmware version request)

omersiar, 2017-04-21:
I can confirm this, i believe this is something about polling SPI bus too much. Reducing

 ```
if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}
```

polling intervals not completely solves the problem but reduces soft resets. We need ESP SPI specifics.

omersiar, 2017-05-12:
`  //If a new PICC placed to RFID reader continue
  if ( ! mfrc522.PICC_IsNewCardPresent()) {
    delay(50);
    return;
  }
  //Since a PICC placed get Serial (UID) and continue
  if ( ! mfrc522.PICC_ReadCardSerial()) {
    delay(50);
    return;
  }`

I made some testings, 50ms delay is enough. My esp never resetted for days.

mmmmar, 2017-05-27:
thanks for your help, I final find the reason is I use two power for esp and mfrc522, and not put thier GND together.  the delay is also neccessary.

closed 2017-05-27 by mmmmar