miguelbalboa/rfid #310: Galileo gen2 - incompatible arch

tomatensaft, 2017-04-27:
### Step 1: Are you in the right place?
GitHub issues only for **bugs** or **improvements** of the library. All other topics, e.g. hardware/programming, will be closed.

For general support from the community, see [Arduino Forum](https://forum.arduino.cc/) or [StackOverflow](https://stackoverflow.com/questions/tagged/mifare).

### Step 2: Describe your environment

 * OS version: win10
 * Arduino IDE version: 1.8.2
 * MFRC522 Library version: 1.3.5
 * Arduino device: galileo gen2
 * MFRC522 device: rc522
  
### Step 3: Describe the problem
cant complie the sketch - incompatible architecture

#### Affected file(s) or example(s):
  
#### Steps to reproduce:
  
#### Observed Results:

 Arduino: 1.8.2 (Windows 10), Board: "Intel® Galileo Gen2"

Build options changed, rebuilding all
WARNING: library rfidmaster claims to run on (avr, STM32F1, teensy, esp8266) architecture(s) and may be incompatible with your current board which runs on (i586) architecture(s).
C:\Program Files (x86)\Arduino\libraries\rfidmaster\src\MFRC522.cpp: In constructor 'MFRC522::MFRC522()':
  
#### Expected Results:
  
#### Relevant Code:
Standard Skecth (Setup/Loop) with #include <MFRC522.h>

What i'm doing wrong ? I found many exmaples with galileo gen2 and MFRC522 !?

Rotzbua, 2017-04-27:
There is no official support for galileo boards. It should compile because it is only a `warning` and not a `error`. Maybe the intel compiler is stupid.. 
Try to remove library.properties where the architectures are listed.

tomatensaft, 2017-04-28:
After many times new installation i got the following log ->

Arduino: 1.8.2 (Windows 10), Board: "Intel® Galileo Gen2"

Build options changed, rebuilding all
WARNING: library MFRC522 claims to run on (avr, STM32F1, teensy, esp8266) architecture(s) and may be incompatible with your current board which runs on (i586) architecture(s).
C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp: In constructor 'MFRC522::MFRC522()':

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:16:33: error: 'UINT8_MAX' was not declared in this scope

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PCD_WriteRegister(MFRC522::PCD_Register, byte)':

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:49:6: error: 'class SPIClass' has no member named 'beginTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:49:72: error: 'SPISettings' was not declared in this scope

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:54:6: error: 'class SPIClass' has no member named 'endTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PCD_WriteRegister(MFRC522::PCD_Register, byte, byte*)':

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:65:6: error: 'class SPIClass' has no member named 'beginTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:65:72: error: 'SPISettings' was not declared in this scope

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:72:6: error: 'class SPIClass' has no member named 'endTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp: In member function 'byte MFRC522::PCD_ReadRegister(MFRC522::PCD_Register)':

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:82:6: error: 'class SPIClass' has no member named 'beginTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:82:72: error: 'SPISettings' was not declared in this scope

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:87:6: error: 'class SPIClass' has no member named 'endTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PCD_ReadRegister(MFRC522::PCD_Register, byte, byte*, byte)':

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:106:6: error: 'class SPIClass' has no member named 'beginTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:106:72: error: 'SPISettings' was not declared in this scope

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:125:6: error: 'class SPIClass' has no member named 'endTransaction'

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PCD_Init()':

C:\data\dev\arduino\sketchbook\libraries\MFRC522\src\MFRC522.cpp:201:28: error: 'UINT8_MAX' was not declared in this scope

exit status 1
Error compiling for board Intel® Galileo Gen2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

When i select the Yun Board it compiles everthing without warnings/errors.

I installed the IDE/Libs/Board new.

Rotzbua, 2017-04-29:
Better open a thread in the arduino or intel forum. Maybe there are more ppl who know about this board.
https://forum.arduino.cc
https://communities.intel.com/community/tech/galileo

Rotzbua, 2017-05-28:
Can be reopened if somebody works on it.

closed 2017-05-28 by Rotzbua

Rotzbua, 2017-06-19:
Intel officially dropped this board https://qdms.intel.com/dm/i.aspx/F79ED317-6EF3-4293-8752-9C4E7B3C11C5/PCN115581-00.pdf .