miguelbalboa/rfid #311: [Lib] Changes to next version

Rotzbua, 2017-05-05:
### Planned

Remove:
* ```#define MFRC522_SPICLOCK SPI_CLOCK_DIV4 // MFRC522 accept upto 10MHz```
* ```MFRC522(byte resetPowerDownPin);``` (breaking)
* ```void PCD_Init(byte resetPowerDownPin);``` (breaking)
* ```#define ```

Add:
* injectable ```SPISettings(MFRC522_SPICLOCK, MSBFIRST, SPI_MODE0)```

Move:
* extra class ```bool MIFARE_OpenUidBackdoor(bool logErrors);
	bool MIFARE_SetUid(byte *newUid, byte uidSize, bool logErrors);
	bool MIFARE_UnbrickUidSector(bool logErrors);``` (breaking)

Feel free to discuss :)

omersiar, 2017-05-12:
Can we configure Interrupt registers, so anyone can benefit from simple attachInterrupt function.

I should try to code it, it will be good learning.

Rotzbua, 2017-05-12:
@omersiar good idea, something like?:
```c++
void PCD_ActivateInterrupt(byte BitmaskInterruptTriggers);
void PCD_ClearInterruptBit();
```
But I think you have to invest some time to find out how exactly the interrupt mechanism works. The example https://github.com/miguelbalboa/rfid/blob/master/examples/MinimalInterrupt/MinimalInterrupt.ino is good but the fct `activateRec` seems for me like a workaround(?) .

A simple function to attach a pin like `PCD_AttachInterruptPin(byte)` seems difficult because it is not very good to let it work under other architectures, so I would disadvice this solution.

lmmeng, 2017-06-10:
Hi @Rotzbua,
the activateRec() function is the "lighter" way I found to energize the board (antenna). Without it the detection of a new card will not work. It is the meaning of the comment in the loop():
```
  // The receiving block needs regular retriggering (tell the tag it should transmit??)
  // (mfrc522.PCD_WriteRegister(mfrc522.FIFODataReg,mfrc522.PICC_CMD_REQA);)
  activateRec(mfrc522);
```
"Lighter" (and fast) because I haven't implemented any check if the commands are passing through, but was just a quick try to see if it has a chance to work. 
Probably there are better (but more time consuming) way to do it, as said, was just a quick hack.

Best regards,
Liviu

ivankravets, 2017-06-13:
I recommend keeping `library.json`, it is more flexible than `arduino.properties`: support platforms, keywords, and other useful things for @PlatformIO registry.

In any case, if you would like to switch on `arduino.properties`, I can make some changes in the Registry.

closed 2018-10-04 by Rotzbua