miguelbalboa/rfid #313: Can read Firmware version but can't read card

Subhajitdas298, 2017-05-09:
* OS version: Win 10
 * Arduino IDE version: 1.8.2
 * MFRC522 Library version: Latest
 * Arduino device: Uno R3(original)
 * MFRC522 device: MRFC522

#### Affected file(s) or example(s):
  
  * Example : dumpinfo.ino
  
#### Steps to reproduce:

  1. Normal opertation
  
#### Observed Results:

  * Shows(reads) firmware version, but can read cards. just nothing happens.
  
#### Expected Results:

  * Expected full dump.

Rotzbua, 2017-05-09:
Seems the "normal" hardware problems, see https://github.com/miguelbalboa/rfid#troubleshooting .

Subhajitdas298, 2017-05-10:
Yup done.
Thanks.
You can close the issue.
Also I have a query : 
If I want to read  few cards one after another, can you give me proper steps, (examples are not clear). I mean when to halt and stoptcrypto1 etc. Need more details.

closed 2017-05-10 by Subhajitdas298

Rotzbua, 2017-05-10:
> If I want to read few cards one after another, can you give me proper steps, (examples are not clear). I mean when to halt and stoptcrypto1 etc. Need more details.

I agree with you, that the examples are not perfect, but that is the only thing we have. The Wiki never started :( https://github.com/miguelbalboa/rfid/wiki .
May search some tutorials, there are lot.

> I think, if we treat a PICC as an object it would be better. How making a class. I make make it, give a pull request, but I need to understand this library more. Need good documentation.

The documentation of the code is very good, just compare it with the official arduino libraries ;) .
If you have some suggestions to refactor until the next release, please write them to #311 .

Best

HJPM, 2017-06-04:
@Subhajitdas298 
I have the very same issue. How did you solve it?

Subhajitdas298, 2017-06-05:
Rotzbua, already helped me.
"Seems the "normal" hardware problems, see https://github.com/miguelbalboa/rfid#troubleshooting ."

Basically  loose connection. For my experienceconnect the PCD directly to arduino and also use max power in code.