miguelbalboa/rfid #314: include "stdint.h"

4ntoine, 2017-05-10:
### Step 1: Are you in the right place?
GitHub issues only for **bugs** or **improvements** of the library. All other topics, e.g. hardware/programming, will be closed.

For general support from the community, see [Arduino Forum](https://forum.arduino.cc/) or [StackOverflow](https://stackoverflow.com/questions/tagged/mifare).

### Step 2: Describe your environment

 * OS version: _Android____
 * Arduino IDE version: _not applicable__
 * MFRC522 Library version: _1.3.5__
 * Arduino device: _any____
 * MFRC522 device: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  
  * MFRC522.cpp
  * stdint.h
  
#### Steps to reproduce:

  1. compile with g++
  
#### Observed Results:

  * failed to compile
```
//data/data/name.antonsmirnov.android.arduinodroid2/sdk/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -DIDE="ArduinoDroid" -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -I/data/data/name.antonsmirnov.android.arduinodroid2/sdk/hardware/tools/avr/avr/include -I/data/data/name.antonsmirnov.android.arduinodroid2/sdk/hardware/arduino/cores/arduino -I/storage/emulated/0/ArduinoDroid/libraries/EEPROM/src -I/storage/emulated/0/ArduinoDroid/libraries/MFRC522/src -I/storage/emulated/0/ArduinoDroid/libraries/SPI/src -I/data/data/name.antonsmirnov.android.arduinodroid2/sdk/hardware/arduino/variants/standard /storage/emulated/0/ArduinoDroid/libraries/MFRC522/src/MFRC522.cpp -o /data/data/name.antonsmirnov.android.arduinodroid2/librariesBuild/MFRC522/MFRC522.cpp.o

05-11 01:22:35.126 W/System.err: /storage/emulated/0/ArduinoDroid/libraries/MFRC522/src/MFRC522.cpp: In constructor 'MFRC522::MFRC522()':
05-11 01:22:35.128 W/System.err: /storage/emulated/0/ArduinoDroid/libraries/MFRC522/src/MFRC522.cpp:16:33: error: 'UINT8_MAX' was not declared in this scope
05-11 01:22:35.131 W/System.err:  MFRC522::MFRC522(): MFRC522(SS, UINT8_MAX) { // SS is defined in pins_arduino.h, UINT8_MAX means there is no connection from Arduino to MFRC522's reset and power down input
05-11 01:22:35.133 W/System.err:                                  ^
05-11 01:22:35.146 W/System.err: /storage/emulated/0/ArduinoDroid/libraries/MFRC522/src/MFRC522.cpp: In member function 'void MFRC522::PCD_Init()':
05-11 01:22:35.148 W/System.err: /storage/emulated/0/ArduinoDroid/libraries/MFRC522/src/MFRC522.cpp:201:28: error: 'UINT8_MAX' was not declared in this scope
05-11 01:22:35.150 W/System.err:   if (_resetPowerDownPin != UINT8_MAX) {
05-11 01:22:35.151 W/System.err:                             ^

```
  
#### Expected Results:

  * Should compile cpp file
  
#### Relevant Code:

```arduino
  // TODO(you): code here to reproduce the problem
```
why is stdint.h included instead of "cstdint.h". it causes incompatibilities for compilation by IDEs
http://stackoverflow.com/questions/13642827/cstdint-vs-stdint-h

4ntoine, 2017-05-10:
For some reason it works in Arduino IDE (but should not work in this way):
```
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_ROBOT_CONTROL -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8038 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Robot Control"' "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/robot_control" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM/src" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src" "-I/Users/asmirnov/Documents/Arduino/libraries/MFRC522/src" "/Users/asmirnov/Documents/Arduino/libraries/MFRC522/src/MFRC522.cpp" -o "/dev/null"
```

Rotzbua, 2017-05-10:
`c++11` do not seem to be enabled by your compile arguments or is it enabled by default?

4ntoine, 2017-05-10:
after adding `-std=gnu++11` it works now. but it still seems to be strange to use `stdint.h` in c++ code

Rotzbua, 2017-05-10:
Maybe strange but not wrong to use `stdint.h`..

closed 2017-05-12 by Rotzbua