miguelbalboa/rfid #315: Availability/Compatibility with Energia 'Platform'?

DrLou, 2017-05-21:
Certainly not a bug... !

But was wondering how difficult it might be ( for me? ) to use this very great library in our Energia work? I see that it's not currently available in energia library repos, but...

Perhaps it's already nearly usable in its current form?

lmmeng, 2017-05-25:
You may want to check the [43oh](http://forum.43oh.com/topic/4028-energia-library-nfc-card-reading-with-energia-mf-rc-522/?do=findComment&comment=76056) forums.

DrLou, 2017-05-25:
Ah, yes. Very good, and thanks for that. ( Ultimately, I was wondering if it had been 'packaged' already as a lib for Energia. )

Rotzbua, 2017-05-28:
@DrLou Hi, can I close this?

DrLou, 2017-06-09:
Immeng: Thanks for the feedback on this. It does suggest a path forward, but actually opens up whole new lines of work. It's not quite a _packaged_ lib yet...

Rotzbua, 2017-06-09:
@DrLou As I understand, the energia platform tries to be compatible to arduino (api) platform. So in theory it should work. The forum mention a spi problem, which point to https://github.com/energia/msp430-lg-core/issues/17 .
I do not exactly understand why are you asking. If you already own a energia board try it. If you do not own a board, only try to compile. Otherwise just buy an arduino ;).

closed 2017-06-09 by Rotzbua