miguelbalboa/rfid #317: Immediately Re-reading a Tag in the event of corrupt data being detected....how?

thorntonforce, 2017-06-02:
* OS version: Windows 10
 * Arduino device: Micro Pro
 * MFRC522 device: MFRC-522 RC522 Kits S50 13.56 
  
I am currently reading 16 bits of Data from tags using mfrc522.PICC_IsNewCardPresent() and mfrc522.PICC_ReadCardSerial() to detect when new tags are present. 

At the end of the function I call     mfrc522.PICC_HaltA(); and mfrc522.PCD_StopCrypto1();

Everything works great but I have noticed that very occasionally (perhaps every 30 scans) there are errors in the 16 bits of data I receive from the tags. This is understandable and acceptable. I implemented a checksum on the data and throw away corrupt packets.

When corrupt packets are detected though it is necessary for the user to momenterilly remove the tag and replace it on the receiver before it is read again.

This is not ideal - what I would like is for in the event of a detected error in the data is for the card to be immediately read again. 

I cant work out how to tell the reader to try again on the same tag in the event of an error occurring?

Can you help?

Thanks

conoror, 2017-06-03:
Funny. I'm hitting this now with my own code (PIC assembler, not a fork of this tree but I do reference it just to get ideas :-) )

What might help you more is the state diagram on Page 12 of the [14443-3 specification](http://wg8.de/wg8n1496_17n3613_Ballot_FCD14443-3.pdf). I get errors sometimes during the effort to move from IDLE to READY (returns 16 bit ATQA). In theory, the card should be READY but might be back at IDLE. 

Thus I'm not sure whether all cards present are READY. Maybe one is IDLE. In which case sending another REQA will put it to READY. And the READY cards back to IDLE. Which isn't very useful at all.

From the state diagram, you can issue a HLTA which will make all cards that aren't ACTIVE go to state IDLE. Then reissue the REQA again. Any card that was ACTIVE will go to HALT but you should know if a card has become active anyway.

conoror, 2017-06-03:
Hope you got all that and aren't more confused. I'll take a look at the code if you have questions.

thorntonforce, 2017-06-03:
Cheers, I did look at the state diagram and figured out a solution. Its not great perhaps but it works.
PICC_IsNewCardPresent() eventually calls PICC_REQA_or_WUPA() with argument PICC_CMD_REQA. This tells the reader to change the state of cards in state "Idle" to state "ready". My code always puts the card into the halt state regardless of whether I have a read error or not; so the tag doesn't get read again until its taken out of the RF field....which presumably puts it back into state Idle.

If you modify PICC_REQA_or_WUPA() and pass the argument PICC_CMD_WUPA instead of PICC_CMD_REQA it invites Tags in state "Idle" AND "HALT"... to go to state "ready".

This means the card is continually read whenever it is in the field.
This actually works for me since I have a function that checks if the ID of the card has already been read and ignores it if it has.

But to improve things further I added a Boolean argument "error" to IsNewCardPresent() to define whether a read error has just occurred. If true it uses PICC_CMD_WUPA to look for halted Tags....if false it uses PICC_CMD_REQA and just looks for Idle tags.

conoror, 2017-06-03:
Yes, that would work. If you send HALT regardless then cards in IDLE or READY will go to IDLE and cards in ACTIVE will go to HALT. At which point WUPA will bring them all to READY regardless. Which is somewhere you can get data from the card instead of it sitting in HALT all the time.

That wouldn't quite work for me as I don't want cards hopping back to Ready all the time - as you say the card will be continuously read whenever it is in the field. But it's great it works for you :-)

Rotzbua, 2017-12-03:
@thorntonforce Can I close this issue?

thorntonforce, 2017-12-03:
Yes...thanks

closed 2017-12-03 by thorntonforce