miguelbalboa/rfid #318: Mifare 1k always gives write timeout

KALRONG, 2017-06-04:
### Step 1: Are you in the right place?
I think this may be a bug in the write function.

### Step 2: Describe your environment

 * OS version: Debian 9 and Windows 10
 * Arduino IDE version: 1.8.1
 * MFRC522 Library version: 2.3.5
 * Arduino device: NANO
 * MFRC522 device: RFID-RC522 firmware 2.0
  
### Step 3: Describe the problem

ChangeUID, FixBrickedUID, RFIDCloner, .... 

Basically all that try to write to the card.
  
#### Steps to reproduce:

 Load any of the previous examples with logs enabled.
  
#### Observed Results:

  All steps seems to works except the write function that timesout almost instantly.
  
#### Expected Results:

  Data to be written in the card.
  
#### Relevant Code:

#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN   9     // Configurable, see typical pin layout above
#define SS_PIN    10    // Configurable, see typical pin layout above

MFRC522 mfrc522(SS_PIN, RST_PIN);   // Create MFRC522 instance

MFRC522::MIFARE_Key key;

void setup() {
  Serial.begin(9600);  // Initialize serial communications with the PC
  while (!Serial);     // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
  SPI.begin();         // Init SPI bus
  mfrc522.PCD_Init();  // Init MFRC522 card
  mfrc522.PCD_SetAntennaGain(0x07<<4);
  Serial.println(F("Warning: this example clears your mifare UID, use with care!"));
  
  // Prepare key - all keys are set to FFFFFFFFFFFFh at chip delivery from the factory.
  for (byte i = 0; i < 6; i++) {
    key.keyByte[i] = 0xFF;
  }
}

void loop() {
  if ( ! mfrc522.PICC_IsNewCardPresent() || ! mfrc522.PICC_ReadCardSerial() ) {
    delay(50);
    return;
  }
  
  // Now a card is selected. The UID and SAK is in mfrc522.uid.
  
  // Dump UID
  Serial.print(F("Card UID:"));
  for (byte i = 0; i < mfrc522.uid.size; i++) {
    Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
    Serial.print(mfrc522.uid.uidByte[i], HEX);
  } 
  Serial.println();
  if ( mfrc522.MIFARE_UnbrickUidSector(true) ) {
    Serial.println(F("Cleared sector 0, set UID to 1234. Card should be responsive again now."));
  }
  delay(1000);
}

Rotzbua, 2017-06-04:
Sector 0 is not a normal sector and normally no writeable. Try a normal data sector.

KALRONG, 2017-06-04:
Already tried with the rfid cloner and have the same timeout error.

Rotzbua, 2017-06-04:
Uff... yes as suggested, than this also writes to sector 0. Read the comment
```
 * Example sketch/program which will try the most used default keys listed in 
 * https://code.google.com/p/mfcuk/wiki/MifareClassicDefaultKeys to dump the
 * block 0 of a MIFARE RFID card using a RFID-RC522 reader.

KALRONG, 2017-06-05:
Really sorry about that, I have tried with both the readandwritte and the rifd_write_personal_data with the same result of timeout out.

Rotzbua, 2017-06-06:
Ok :D , did you tried to increase antenna gain?

KALRONG, 2017-06-06:
Yep, three diferent tags and if you check the code I pasted you can see the line:
mfrc522.PCD_SetAntennaGain(0x07<<4);
Right after the init, thats what I found around that it should be done to power the antenna to the max.

Rotzbua, 2017-08-19:
@KALRONG  Hi, did you found a solution? Or tried another mfrc522 module?

KALRONG, 2017-08-21:
Hi, sorry for the long time without updates, but its been quite a busy time for me.

I couldn't try with another mfrc522 module but my guess is that the cards I got are just crap, some of them doesn't even read let alone write to them :( The chinese who sold them to me actually said that the module wasn't a writter ....

Feel free to close this issue if you want, I will try to get my hands in a new set of cards and another mfrc522 module just in case.and hopefully I will have more luck hehe

Thanks a lot for your help!

closed 2017-08-21 by Rotzbua