miguelbalboa/rfid #319: RFID-Reader is connected but wont read any tags

HJPM, 2017-06-04:
Hey all,
I am trying to get my RFID-reader working but im at a loss now.
I am quite sure i connected it right. If i start the "DumpInfo" example it shows me on the monitor:

Firmware Version: 0x92 = v2.0
Scan PICC to see UID, SAK, type, and data blocks...

So I think im connected and there is some sort of communication running between arduino and RFID-reader.
I checked the connections may times now, resoldered my pins, using a step 3.3V to 5V step down converter and tried 2 different readers with 5 different tags (they all came with the reader, so i think they are the right tags). All with the same result:
The monitor shows no changes at all if i move a tag in the vicinity of the reader.
Anybody got any ideas? Is there a known conflict with the new versions of whatever?

Hope someone has any idear about what else to try.

Thanks so far
HJPM

### Step 2: Describe your environment

 * OS version: WIN10
 * Arduino IDE version: 1.8.2
 * MFRC522 Library version: 1.3.6
 * Arduino device: UNO
 * MFRC522 device: https://www.amazon.de/MFRC-522-Module-KeyCard-Arduino-Raspberry/dp/B071H9NLS5/ref=sr_1_4?ie=UTF8&qid=1496609872&sr=8-4&keywords=arduino+rfid

Rotzbua, 2017-06-04:
If you get `Firmware Version: 0x92 = v2.0
Scan PICC to see UID, SAK, type, and data blocks...`, the reader is correct connected.
So may try my list: https://github.com/miguelbalboa/rfid#troubleshooting
Otherwise you received bad hardware and you have to contact your seller. Best.

HJPM, 2017-06-10:
I ordere two new ones form a different shop.
They worked well on the first time. So i guess the hardware was broken.
Thanks for the help.

Wissenss, 2023-05-11:
Hey, I just solve this same problem. In my case it wasn't that the hardware was broken but the way I was testing it. 
You see I place my chip facing towards the conexion cables, this created interference and thus stop my chip from properly detecting the tag, just face it the other way and it works! Hope this helps someone :b

RRR3d, 2023-12-23:
@Wissenss  This fixed my  problem, Thanks bro :)