miguelbalboa/rfid #321: add problem

giumbyyyyy, 2017-06-11:
### Step 1: Are you in the right place?
GitHub issues only for **bugs** or **improvements** of the library. All other topics, e.g. hardware/programming, will be closed.

For general support from the community, see [Arduino Forum](https://forum.arduino.cc/) or [StackOverflow](https://stackoverflow.com/questions/tagged/mifare).

### Step 2: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```arduino
  // TODO(you): code here to reproduce the problem
```

giumbyyyyy, 2017-06-11:
Hello. I've been trying to add more than 64 cards. At address 257 in the eeprom, resume the count from 1. Why?

giumbyyyyy, 2017-06-11:
Acces control sketch

Rotzbua, 2017-06-12:
Nice try, simple math solves the problem: 16 byte * 64 = 1024 byte.

closed 2017-06-12 by Rotzbua