miguelbalboa/rfid #323: How to solve "A BUFFER IS NOT BIG ENOUGH" error ?

AyushMarsian, 2017-06-14:
Hello Friends,

I m using an mfrc522 for one project.

The issue is; Normally my MRRC522 works well, but if we remove card from MFRC522 while operation is going on then some error occur like "Authentication fail","Timeout in Communication" then it will recover back.

But when it will shows **"A BUFFER IS NOT BIG ENOUGH"** then it will not recover back only shows this error every time.

I have tried mfrc522.PCD_Reset(); and mfrc522.PCD_Init(); function of Library but is not work, i have to compulsory restart the board altogether.

How can the crc error be cleared without resetting/powering up the board ?

Appreciate all the help.

Thanks

Rotzbua, 2017-06-14:
There is a reason for the github issue template. Extend your issue with your environmental information, otherwise I will close this. Thanks.

AyushMarsian, 2017-06-14:
I'm sorry for inconvenience, but I'm new in GitHub,

I don't understand what you want to say please describe me in detail

Thanks

Rotzbua, 2017-06-14:
If you open a issue this will be shown: https://github.com/miguelbalboa/rfid/blob/master/.github/ISSUE_TEMPLATE.md
`Step 2` is missing and an example/code would be good. Just edit your first post. Thanks.

Freebornrus, 2017-07-23:
1. What if you increase buffer size from 18 to...64, 128? Just to check.
2. How many attempts are on your "Observed Results"?

AyushMarsian, 2017-08-05:
i don't know where is the buffer so i can't increase it.

and i was observed results greater then 50 times.

Freebornrus, 2017-08-05:
@AyushMarsian 
Up in the code 
`byte buffer[18]     ={0};`

AyushMarsian, 2017-08-05:
@Freebornrus 
Thanks , 
But it's for data bytes from Card and data byte of card is fix 16 data bytes and 2 checksum bytes so it's of 18 bytes (See in image). 
it will never increase or nor decrease then 18, so i don't think it's creating issue.

[image]

Rotzbua, 2017-08-17:
@AyushMarsian Did you solved this problem?

Read the documentation: https://github.com/miguelbalboa/rfid/blob/1.3.6/src/MFRC522.cpp#L899
I think then you can solve your problem by your self ;) (hint: value of the pointer `&size`)

AyushMarsian, 2017-08-17:
Thanks @Rotzbua 

Could you tell me when this error will occur "MIFARE_Read() failed: A buffer is not big enough." Because that message from the library, so what part of library i have to modify (value of the pointer &size) ?

I have studied .cpp file of library and first i thought that RC522 is hanged but that error is of library and can't recover so i have to reset controller.

currently i'm continue with this 

status = (MFRC522::StatusCode) mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, trailerBlock, &key, &(mfrc522.uid));
      if (status != MFRC522::STATUS_OK)
      {
        Serial.print(" Error code ");
        Serial.println(mfrc522.GetStatusCodeName(status));
        if(status==4) // status code 4 means MIFARE_Read() failed: A buffer is not big enough
        {
          wdt_enable(WDTO_15MS);
          delay(30);
        }
      }

please help me

Rotzbua, 2017-08-17:
The documentation tells you, that a call of `mfrc522.MIFARE_Read(blockAddr, buffer, &size)` change the value of `size`.

>   byte *bufferSize	///< Buffer size, at least 18 bytes. Also number of bytes returned if STATUS_OK.
) {

It seems that the documentation is not completely accurate and the value is changed also if STATUS_?? . Maybe somebody can have a closer look. Idk...

So if you call again  `mfrc522.MIFARE_Read(blockAddr, buffer, &size)`, `size` would maybe contain `size=0`. Means your buffer is too small and function fails.
Conclusion: after you call `mfrc522.MIFARE_Read(blockAddr, buffer, &size)`, have to reset `size` with `size= sizeof(buffer);`.

Best

AyushMarsian, 2017-08-17:
Thank you so much @Rotzbua 
I'll try this and come back

closed 2017-08-19 by AyushMarsian

AyushMarsian, 2017-08-20:
I Solve my Problem by adding ```size= sizeof(buffer);``` in loop 

Thank you @Rotzbua @Freebornrus

philippejadin, 2018-05-12:
@AyushMarsian could you post yout working code? I still can't manage to make it work without watchdog (which is not a bad idea anyway). Except if I remove the "return" from the status reporting block. I don't fully understand the impliction of a return in the context of loop() (for garbage collection, and proper shutdown of rfid lib)

thank you in advance!

AyushMarsian, 2018-05-15:
@philippejadin Visit following URL for Download Arduino Code
[https://drive.google.com/open?id=1xYrcjoMfyNnwBfhA87w_-TNlJAarJDTX](url)
Can you explain me your problem, when you use watchdog for reset controller ?
For more query related to this mail me on ankitghevariya358@gmail.com

philippejadin, 2018-05-18:
What works for me is to stop the card before returning : 
```

  status = mfrc522.MIFARE_Write(block, &data[16],  16);
  if (status != MFRC522::STATUS_OK) {
    Serial.print(F("MIFARE_Write() failed: "));
    Serial.println(mfrc522.GetStatusCodeName(status));

    delay(100);
    mfrc522.PICC_HaltA();
    mfrc522.PCD_StopCrypto1();
    return false;
}
```

parthkarnawat, 2018-12-31:
> What works for me is to stop the card before returning :
>
> ```
>
>   status = mfrc522.MIFARE_Write(block, &data[16],  16);
>   if (status != MFRC522::STATUS_OK) {
>     Serial.print(F("MIFARE_Write() [...]

Hey @philippejadin  the 'The CRC_A does not match' error has nothing to do with the buffer size it means that the crc coproccesor on the MFRC522 chips calculations might have gone wrong(for some reasons that I don't know, maybe manufacturing defect after all it is a very cheap chip).
Though you can get over the problem without using the watchdog by using PCD_Reset() and PCD_init()

`status = mfrc522.MIFARE_Read(block, buffer2, &len);
  if (status != MFRC522::STATUS_OK) {
    Serial.print(F("Reading failed: "));
    Serial.println(mfrc522.GetStatusCodeName(status));
    mfrc522.PCD_Reset();
    mfrc522.PCD_Init();
    return;
  }
`
So now you don't have to reset the whole board !

matonga, 2019-09-05:
Hi, just wanted to say I had a similar problem. From time to time the MFRC522 module would not read cards anymore.

The solution was to move PCD_StopCrypto1. I was calling it after reading authenticated sector. Now I call PCD_StopCrypto1 **before** calling PICC_IsNewCardPresent. Yes, every time I call PICC_IsNewCardPresent, I call PCD_StopCrypto1 before, just in case.
```
while (1) {
  PCD_StopCrypto1 ( );
  if (PICC_IsNewCardPresent ( )) {
    // select the card, authenticate, read block, etc... halt the card
    // ...
  }
}
```
I left an automated rig passing four cards all night, and the next morning it was still reading the cards successfully. Hope this helps someone else because this library is great.

Shinytastic, 2019-10-08:
> > What works for me is to stop the card before returning :
> > ```
> > 
> >   status = mfrc522.MIFARE_Write(block, &data[16],  16);
> >   if (status != MFRC522::STATUS_OK) {
> >     Serial.print(F("MIFARE_Write() failed: "));
> >     Serial.println(mfrc522.GetStatusCodeName(status));
> > 
> >     delay(100);
> >     mfrc522.PICC_HaltA();
> >     mfrc522.PCD_StopCrypto1();
> >     return false;
> > }
> > ```
> 
> Hey @philippejadin & @AyushMarsian the 'The CRC_A does not match' error has nothing to do with the buffer size it means that the crc coproccesor on the MFRC522 chip must have found some error in the data(for some reasons that I don't know, maybe manufacturing defect after all it is a very cheap chip).
> check this out 'https://en.wikipedia.org/wiki/Cyclic_redundancy_check'
> Though you can get over the problem without using the watchdog by using PCD_Reset() and PCD_init()
> 
> `status = mfrc522.MIFARE_Read(block, buffer1, &len); if (status != MFRC522::STATUS_OK) { Serial.print(F("Reading failed: ")); Serial.println(mfrc522.GetStatusCodeName(status)); if(status==7) // status code 7 means the crc_a didn't match { mfrc522.PCD_Reset(); mfrc522.PCD_Init(); } return; } `
> So now you don't have to reset the whole board !

Massive thanks for this. I think you just saved me losing half my day.

allecsxxx, 2021-02-16:
Sry may be a late reply:

mfrc522.cpp

Search for this : if (n > *backLen) {

                
			return STATUS_NO_ROOM;

		}

Modify like this : if (n > *backLen) {

                *backLen = n;
			return STATUS_NO_ROOM;

		}

Every time it gets in this if statement it will reset FIFO register.

Rotzbua, 2021-02-16:
@allecsxxx Is this a fix for a bug or just an individual solution? If a fix then it would be nice to see a PR.

allecsxxx, 2021-02-18:
> @allecsxxx Is this a fix for a bug or just an individual solution? If a fix then it would be nice to see a PR.

This is not a bug or so.... it is a missing bit from library. When it gets to that if statement  it will return No_ROOM and before this If statement there is nothing that alters the   backLen  value. So if you get this if statement once , program will always end there. If you look in .cpp after this if statement backLen gets n value, but too late as program ended before. 

This a buffer is not big enough happens when fast card swipe is applied and comunication gets interrupted. 
Basically my modification in if statement is to make next swipe of a card after "a buffer is not big enough " error to work.

allecsxxx, 2021-02-18:
I m playing with this module and library  and i encountered same issue and when it happens,  it is totally right to happen otherwise you might get incorrect data back from the card. But if it happens once doesn't mean everything must stop and restart to be aplied, just need a clear.

philippejadin, 2021-02-18:
I think in this case a pull request would be a good idea, since the "fast card swipe" that stops the reader from working is indeed a big problem.

allecsxxx, 2021-02-18:
I quickly went again through all the lines related to issue A BUFFER IS NOT BIG ENOUGH. I could not find anything that will change "n" variable once the program gets to A BUFFER IS NOT BIG ENOUGH error, n variable gets changed after the if statement as per original library so it is enough to get into that if statement once and program will always return from there, not continuing..so because of this my suggestion above is to accept the error A BUFFER IS BIG ENOUGH and clear "n " variable so program can run again and read cards again without restarting arduino. The library does a good job , when it shows the error it is 100% right and it happens because communication is interrupted too early and it gets buggy (swiping cards too fast, fast as a rocket passing above the antenna).  I think the creator of this library missed this part and thought no one swipes 30 40 times with the speed of sound same card in front of the reader.

jitendra1607, 2021-08-30:
Use below code to continue reading 

`status = mfrc522.MIFARE_Read(block, buffer2, &len);
  if (status != MFRC522::STATUS_OK)
  {
    Serial.print(F("Reading failed status code: "));
    Serial.println(status);

      mfrc522.PCD_SoftPowerDown();
    delay(500);
    mfrc522.PCD_SoftPowerUp();
    mfrc522.PCD_Init(ssPin, RST_PIN); // re-init
}
`

AyushMarsian, 2021-08-30:
@jitendra1607 

Well, to reset/restart the RFID/NFC scanner is not a proper solution. And in the above case there was not a problem of RFID Scanner. The actual problem was in the size of buffer which we used to collect the data frame in controller. So, my issue only get solve when I reset the MCU of Arduino. Now, when I realized that there is problem in the variable which stores size, I just refill it every time before read or write operation.

For more look at this https://github.com/miguelbalboa/rfid/issues/323#issuecomment-323072804

Thank you

matonga, 2021-08-30:
My brain's RAM could be failing, but I recall supplying a 16 byte buffer to the library and having this problem, because the library expected an 18 byte buffer to add a 2 byte CRC or something.

jitendra1607, 2021-08-31:
@AyushMarsian Please note that size= sizeof(buffer); is already there in library, so no need to do it in code.

 ```
  byte block;
  byte len;
  MFRC522::StatusCode status;

  len = 18;
  byte buffer2[18];
  block = 1;

  status = mfrc522.PCD_Authenticate(MFRC522::PICC_CMD_MF_AUTH_KEY_A, 1, &key, &(mfrc522.uid));
  status = mfrc522.MIFARE_Read(block, buffer2, &len);
```

auduner, 2023-02-26:
> Hi, just wanted to say I had a similar problem. From time to time the MFRC522 module would not read cards anymore.
>
> The solution was to move PCD_StopCrypto1. I was calling it after reading [...]

Super thanks for this!!! I almost gave up. This solved all my troubles.

NaderKheirDT, 2024-07-06:
put this in the void loop and remove it from global variable it will solve the issue 
byte buffer[18]     ={0};

NaderKheirDT, 2024-07-08:
you can change the following in the cpp file 

	if (bufferATQA == nullptr || *bufferSize < 2) {	// The ATQA response is 2 bytes long.
	    // Reset the buffer pointer
        bufferATQA = new byte[18]; // Allocate new memory for the buffer
        *bufferSize = 18;

    
		return STATUS_NO_ROOM;
	}

and

"	if (buffer == nullptr || *bufferSize < 18) {
		    // Reset the buffer pointer
        buffer = new byte[18]; // Allocate new memory for the buffer
        *bufferSize = 18;
    
		return STATUS_NO_ROOM;
	}"